siemens/mtda #11: integrate with LAVA

chombourger, 2020-10-09:


hkoturap, 2020-10-27:
@chombourger ,
Will the Assist Board become the DUT for LAVA and complete target board control will be with Assist Board.
In this scenario updating the test suites result in LAVA when executed in Target board may be issue.

chombourger, 2020-10-27:
the use case is somewhat similar to https://github.com/bootlin/lavabo
we may want LAVA to go through MTDA to power the board ON/OFF, write the OS image and get console output

chombourger, 2020-11-01:
lava seems to support `to: flasher` which I believe could be used to issue `mtda storage commands`

chombourger, 2020-11-01:
see https://staging.validation.linaro.org/scheduler/device/staging-soca9-01/devicedict and https://staging.validation.linaro.org/scheduler/job/262229/definition

chombourger, 2020-11-04:
documentation added. available from f7ce42b5f45b7088fe4f467c55b1da04e8c925ca and later

closed 2020-11-04 by chombourger