siemens/mtda #4: configuration option to change prefix key

chombourger, 2020-10-09:
MTDA may be controlled from the interactive console by using a key combination of a prefix key, `Ctrl-a` by default, followed by a command key. People using a terminal multiplexer such as `screen` may find our default prefix key conflicting with the multiplexer. Provide a configuration option to change the default key prefix to any letter (e.g. `Ctrl-m`

hkoturap, 2020-10-20:
@chombourger , 
Should the configuration option to change the default key prefix be provided in mtda.ini. Should all letters be taken care for configuration option(http://ascii-table.com/control-chars.php)

chombourger, 2020-10-20:
@hkoturap let's start with letters and yes this would go in mtda.ini

closed 2020-10-27 by chombourger