siemens/mtda #29: need a logo for this project

chombourger, 2020-11-05:


chombourger, 2021-12-30:
> [image]

I like the design and the explanation. maybe we just add some colors to it?

chombourger, 2021-12-30:
maybe we just add the various controls underneath
[image]
(this is just a quick-n-dirty mock-up)

vj-kumar, 2022-01-04:
To note, the Bluetooth logo is a registered trademark, and we cannot possibly use it in a logo I guess. There are certain guidelines on how it can be used. Also, one needs to be part of Bluetooth Sig.

https://www.bluetooth.com/develop-with-bluetooth/marketing-branding/
https://www.bluetooth.com/wp-content/uploads/2021/01/Bluetooth-Brand-Guides-and-Logos-English.zip

vj-kumar, 2022-01-04:
I could try to colorize the logo. Not an expert in GIMP/Photoshop. Might take some time though.

GuBi34, 2022-05-02:
I tried to copy some catchy graphical representance of the letters themselves. Probably you like it .. if so I'd make a 3D model and render it as
a) icon
b) animated 3d model
c) ???

[image]