siemens/mtda #40: add a wifi driver

chombourger, 2020-11-26:
Some test suites require having a Wi-Fi access point to exercise their 802.11 stack. We could plug a USB Wi-Fi dongle to our NanoPI and provide APIs (most likely simple wrappers around NetworkManager calls) and configure it as an Access Point.

vj-kumar, 2022-01-12:
If I understand correcly, We need to be able to configure the WiFi over mtda(as AP) so that some wifi test cases(perf/ping) can use that instead of having the need to have some external wifi hardware.

chombourger, 2022-01-12:
> If I understand correcly, We need to be able to configure the WiFi over mtda(as AP) so that some wifi test cases(perf/ping) can use that instead of having the need to have some external wifi hardware.

the use-case is that some devices to be tested will have Wi-Fi capabilities. We would like the board assist (like the Nanopi R1) to gain Wi-Fi activities. We could then write tests to configure the NanoPI as an Access Point, set a specific PSK, deliver a DHCP address, etc.

Some potential `mtda-cli` calls:
```
mtda-cli wifi mode ap
mtda-cli wifi ssid test-network
mtda-cli wifi security WPA2/PSK
mtda-cli wifi psk HackMe!
mtda-cli wifi on
```
we may also want to have a `dhcp` interface so we could enable/configure a DHCP server on the nanopi