siemens/mtda #97: unable to install prebuilt packages on Ubuntu 20.04

chombourger, 2021-12-06:
Reported by @bhargavdas 

after adding fury.io/mtda to source list and then while installing  mtda getting dependency errors,
mainly for python3-zstandard 

`python3-zstandard : Depends: python3 (>= 3.9~) but 3.8.2-0ubuntu2 is to be installed`

bhargavdas, 2021-12-06:
This issue is also observed in debian 10 which has `python3.7` installed.

chombourger, 2021-12-06:
> This issue is also observed in debian 10 which has `python3.7` installed.

right. this will be subject of a separate issue (if we find ourselves wanting to support buster)

chombourger, 2021-12-06:
#98 should help

chombourger, 2022-01-20:
0.15 packages at https://launchpad.net/~chombourger/+archive/ubuntu/mtda-focal
They were created with a script that ill be added to our GitHub actions

commit 823a565f (2022-01-21): ci: upload source packages for Ubuntu to PPA

Closes: #97
closed 2022-01-21 by chombourger