siemens/mtda #105: Hardware support from mtda for nanopi R1

GuBi34, 2021-12-10:
Can mtda support the FriendlyArm NanoPi R1?

Rational:

It would be benificial for us to instrument the test device as well with the network environment (dhcp / fixed IP / gateway / proxy ...). This will allow us to call some device APIs from the test dispatcher as well without the need to have a parallel configuration of switches and routers.

As well more than one USB port would be helpful to avoid USB HUBs.

Looking for WiFi devices a local Hot Spot will be a help.

The nanopi R1 seems to fit well. It is thought as minirouter and comes with nice housing options.

https://www.friendlyarm.com/index.php?route=product/product&path=69&product_id=248

chombourger, 2021-12-20:
The Debian kernel does not seem to be shipping `sun8i-h3-nanopi-r1.dts` (at first glance)

chombourger, 2021-12-20:
It does not seem to exist in 5.10 but may be there in 5.11 (see https://elixir.bootlin.com/linux/v5.11/source/arch/arm/boot/dts/sun8i-h3-nanopi-r1.dts)

chombourger, 2021-12-20:
so maybe we could pull the kernel from bullseye-backports so we get 5.14
https://packages.debian.org/bullseye-backports/linux-image-armmp

GuBi34, 2021-12-20:
I just made it to install FriendlyWRT on the device. Since I have now root login I may scan the hardware for you (possibly).

This is what I see at a first glance

```
uname -a
Linux FriendlyWrt 4.14.111 #1 SMP Tue May 11 18:37:06 CST 2021 armv7l GNU/Linux
```

chombourger, 2021-12-20:
Hallo @GuBi34, See if #112 helps

The image I built still boots on the nanopi NEO but it now comes with a 5.14 kernel and has the following dtb files:

- sun8i-h3-nanopi-duo2.dtb 
- sun8i-h3-nanopi-m1-plus.dtb  
- sun8i-h3-nanopi-neo.dtb
- sun8i-h3-nanopi-m1.dtb    
- sun8i-h3-nanopi-neo-air.dtb 
- sun8i-h3-nanopi-r1.dtb 

The u-boot environment of your r1 should have `fdtfile` set to `sun8i-h3-nanopi-r1.dtb`. The image should hopefully load without needing any tweaks.

I can provide the image I have built if you'd like.

chombourger, 2021-12-22:
assigned this issue to 0.15 since we do not have this hardware yet. We will check if we can purchase some in January (hardware donations are always an option and appreciated, simply reach out if you would like to donate to the project).

chombourger, 2021-12-23:
issue #40 will become relevant as soon as we have support for the NanoPI R1. The agent would receive APIs to control the Wi-Fi interface: e.g. make ourselves an Access Point

chombourger, 2021-12-23:
Various logs provided by Gunther
# Boot `FriendlyElec` (Ubuntu core with U-Boot) on `NanoPi-Neo`

```
U-Boot SPL 2017.11 (Jun 15 2021 - 10:49:36)
DRAM: 512 MiB(408MHz)
CPU Freq: 408MHz
memory test: 1
Pattern 55aa  Writing...Reading...OK
Trying to boot from MMC1
Boot device: sd

U-Boot 2017.11 (Jun 15 2021 - 10:49:36 +0800) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyElec NanoPi H3
DRAM:  512 MiB
CPU Freq: 1008MHz
MMC:   SUNXI SD/MMC: 0, SUNXI SD/MMC: 1
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
BOARD: nanopi-neo
starting USB...
No controllers found
Hit any key to stop autoboot:  0
reading boot.scr
1478 bytes read in 19 ms (75.2 KiB/s)
## Executing script at 43100000
running boot.scr
reading uEnv.txt
969 bytes read in 18 ms (51.8 KiB/s)
reading zImage
5902784 bytes read in 291 ms (19.3 MiB/s)
reading rootfs.cpio.gz
5880768 bytes read in 291 ms (19.3 MiB/s)
reading sun8i-h3-nanopi-neo.dtb
34439 bytes read in 26 ms (1.3 MiB/s)
overlays=uart0
reading overlays/sun8i-h3-uart0.dtbo
508 bytes read in 27 ms (17.6 KiB/s)
applying overlay uart0...
reading overlays/sun8i-h3-fixup.scr
4109 bytes read in 36 ms (111.3 KiB/s)
## Executing script at 44500000
## Flattened Device Tree blob at 48000000
   Booting using the fdt blob at 0x48000000
   Loading Ramdisk to 49a64000, end 49fffbc0 ... OK
   reserving fdt memory region: addr=48000000 size=6e000
   Loading Device Tree to 499f3000, end 49a63fff ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.111 (root@ubuntu) (gcc version 4.9.3 (ctng-1.21.0-229g-FA)) #193 SMP Thu Jun 10 18:20:47 CST 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=30c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: FriendlyElec NanoPi-NEO
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000005bc00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 17 pages/cpu @dff5b000 s37120 r8192 d24320 u69632
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: console=ttyS0,115200 earlyprintk root=/dev/mmcblk0p2 rootfstype=ext4 rw rootwait fsck.repair=yes panic=10 fbcon=map:0  data=/dev/mmcblk0p3 snd-soc-core.pmdown_time=3600000
[... 1777 lines cut ...]
         Starting Permit User Sessions...
         Starting Hostname Service...
[  OK  ] Finished Permit User Sessions.
[  OK  ] Started User Login Management.
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Reached target Login Prompts.
[   18.970074] Mass Storage Function, version: 2009/09/11
[   18.975439] LUN: removable file: (no medium)
[   19.009670] mass_storage.usb0/lun.0: unable to open backing file: /dev/sda
[   19.032720] mass_storage.usb0/lun.0: unable to open backing file: /dev/sda
[  OK  ] Started Hostname Service.
[  OK  ] Listening on Load/Save RF …itch Status /dev/rfkill Watch.
         Starting Network Manager Script Dispatcher Service...
[   19.829264] dwmac-sun8i 1c30000.ethernet eth0: PHY [0.1:01] driver [Generic PHY] (irq=POLL)
[   19.845080] dwmac-sun8i 1c30000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   19.854942] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[   19.862509] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   19.870382] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[  OK  ] Started Network Manager Script Dispatcher Service.
[   19.907149] dwmac-sun8i 1c30000.ethernet eth0: configuring for phy/mii link mode
[  OK  ] Finished Remove Stale Onli…ext4 Metadata Check Snapshots.

Debian GNU/Linux 11 mtda ttyS0
```

chombourger, 2021-12-29:
```
[    4.812367] dwmac-sun8i 1c30000.ethernet: IRQ eth_wake_irq not found
[    4.819555] dwmac-sun8i 1c30000.ethernet: IRQ eth_lpi not found
```
looked suspicious but the error message turns out to be misleading since these are optional IRQ lines
a fix was proposed upstream https://patchwork.ozlabs.org/project/netdev/patch/20200306163848.5910-1-mklntf@gmail.com/

chombourger, 2021-12-29:
More concerning is
```
Model: FriendlyARM NanoPi NEO
Loading /usr/lib/linux-image-5.14.0-0.bpo.2-armmp/sun8i-h3-nanopi-neo.dtb...
```

chombourger, 2021-12-29:
That was the issue. was able to boot a MTDA image on the r1:
```
root@mtda:/home/mtda# dmesg|grep -i r1
[    0.000000] OF: fdt: Machine model: FriendlyARM NanoPi R1
```
will now prepare the pull request

chombourger, 2021-12-29:
initial support with #123 (thanks @GuBi34 for the board you sent me!)

the image was booted from SD. Need to check if patches are needed to support distro_boot (see https://marcin.juszkiewicz.com.pl/2021/03/14/u-boot-and-generic-distro-boot/): this would let us use the same image for either SD or eMMC boot

several packages to be added to support BT/Wi-Fi
I have not yet checked the second Ethernet port

chombourger, 2021-12-29:
2nd ethernet port is also working:
```
mtda@mtda:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 42:59:38:32:eb:c9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.48/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 83044sec preferred_lft 83044sec
    inet6 2a01:cb1d:84b4:9600:12e4:d56a:baa:b0d7/64 scope global dynamic noprefixroute 
       valid_lft 1762sec preferred_lft 562sec
    inet6 fe80::6a26:f34e:7c09:ea57/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 3e:cf:6b:17:75:44 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.50/24 brd 192.168.1.255 scope global dynamic noprefixroute eth2
       valid_lft 86322sec preferred_lft 86322sec
    inet6 2a01:cb1d:84b4:9600:a59:f214:4c1b:3f72/64 scope global dynamic noprefixroute 
       valid_lft 1762sec preferred_lft 562sec
    inet6 fe80::dbb4:e6e4:e24:3d66/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

vj-kumar, 2021-12-30:
Ah. So the device tree is the issue. I also received my R1. Will look into WiFi tonight.

vj-kumar, 2021-12-30:
linux-firmware package is not present. We would need that.

vj-kumar, 2021-12-30:
Wifi module based on broadcom. So brcmfmac driver should ideally be sufficient with the right firmware. I see the driver. Let me put the linux-firmware package and check.

chombourger, 2021-12-30:
I would suggest we add BT/Wi-Fi support in a separate PR but yes we probably need to add several packages (we also need the package that ships regulatory.db)

chombourger, 2021-12-30:
> Ah. So the device tree is the issue. I also received my R1. Will look into WiFi tonight.

right. the u-boot binary from Debian does not recognize the R1 and somehow thinks it is a NEO. That's how we got the DT for the NEO loaded :(

vj-kumar, 2021-12-30:
> I would suggest we add BT/Wi-Fi support in a separate PR but yes we probably need to add several packages (we also need the package that ships regulatory.db)

Agreed. I have not decoded the dtb yet. Maybe there is wifi entry. Maybe not.

chombourger, 2021-12-30:
Modified the wks to boot Linux with `root=PARTUUID=2edae528-01` instead of `root=/dev/mmcblk1p1`
Verified that the same image may now be used either on the SD card or eMMC

chombourger, 2021-12-30:
Programmed the eMMC as follows:
```
mtda@mtda:~$ sudo dd if=isar-image-base-debian-bullseye-nanopi-r1.wic.img of=/dev/mmcblk2 bs=8M status=progress
```

closed 2021-12-30 by chombourger

GuBi34, 2021-12-30:
@chombourger thanks for the ultra quick support!!!

commit 80dba29f (2021-12-30): meta-isar/conf: add nanopi-r1 machine

Closes: #105