siemens/mtda #107: Cross Compile Issues

michaeladler, 2021-12-17:
I have noticed that this layer fails to build using `ISAR_CROSS_COMPILE = "1"` (which is often enabled globally in downstream layers which may depend on this layer). It seems that mostly Python packages are affected by this, e.g.

* curve25519-donna
* hap-python

(this list is most likely incomplete)

It would be nice if this layer could be built when cross-compilation is enabled globally. There are two viable approaches:

1. Fix the problematic recipes (e.g. by depending on python-whatever:**native**).
2. Disable cross-compilation for recipes which cannot be fixed easily.

chombourger, 2021-12-17:
will check. thanks for reporting this issue.

chombourger, 2021-12-17:
`ed25519` is also impacted

chombourger, 2021-12-17:
@michaeladler here's https://github.com/siemens/mtda/pull/108
going to burn the nanopi image to make sure it is all ok

michaeladler, 2021-12-20:
That's great, thanks! Looking forward to it.

commit 76b3b551 (2021-12-20): meta-isar/recipes-python: fix Build-Depends to support cross-compiling

Fixes: #107
commit a59d9564 (2021-12-20): kas/debian: set ISAR_CROSS_COMPILE="1" for the nano-pi

Fixes: #107
closed 2021-12-20 by chombourger