siemens/mtda #121: support xz compressed images

chombourger, 2021-12-23:
use https://github.com/Rogdham/python-xz to support .xz images

vj-kumar, 2022-01-02:
Not sure if someone is looking into to it. Planing to take it.

chombourger, 2022-01-02:
> Not sure if someone is looking into to it. If not planing to take it.

Thanks!

vj-kumar, 2022-01-02:
@chombourger Any specific reason for using python-xz? Looks like lzma can do it to. I do see some benefits mentioned in python-xz, but currently not sure whether it will have an impact.

jan-kiszka, 2022-01-02:
If you are talking about isar image compression: Adriaan has started porting over the OE image compression features. That will obsolete the currently needed custom image classes.

vj-kumar, 2022-01-02:
@jan-kiszka This is about mtda's storage writer supporting xz compressed images. https://github.com/siemens/mtda/blob/a13e98c44271218adfb850f23119a9788234265b/mtda/storage/writer.py#L49-L60

vj-kumar, 2022-01-08:
Bz2 write is broken. Tested in Qemu, Richa also validated in an actual board. XZ similar implementation to BZ2 also doesnot work. Checking.

chombourger, 2022-01-08:
Thanks for the info. We will get a framework to test our APIs and features With https://github.com/siemens/mtda/pull/146 and hopefully avoid regressions in the future

vj-kumar, 2022-01-10:
Related to https://github.com/siemens/mtda/issues/147

Planning to address #147 before getting here.

PR #153 (2022-01-12): mtda: add support for xz compressed images
Based on bz2 logic.

Closes: #121
commit bb7e05d3 (2022-01-12): mtda: add support for xz compressed images

Based on bz2 logic.

Closes: #121
closed 2022-01-12 by chombourger