siemens/mtda #125: nanopi-r1: enable Wi-Fi

chombourger, 2021-12-30:


chombourger, 2021-12-30:
We would need:

- firmware-brcm80211
- wireless-regdb
- wireless-tools

chombourger, 2021-12-30:
we also need to ship `/lib/firmware/brcm/brcmfmac43430-sdio.txt`
I extracted this file from the images provided by FriendlyArm:
```
# NVRAM file for BCM943430WLSELG (AW-NB197SM)
# 2.4 GHz, 20 MHz BW mode

# The following parameter values are just placeholders, need to be updated.
manfid=0x2d0
prodid=0x0726
vendid=0x14e4
devid=0x43e2
boardtype=0x0726
boardrev=0x1101
boardnum=22
macaddr=00:11:22:33:44:55
sromrev=11
boardflags=0x00404201
xtalfreq=26000
nocrc=1
ag0=255
aa2g=1
ccode=ALL

pa0itssit=0x20
extpagain2g=0

#PA parameters for 2.4GHz, measured at CHIP OUTPUT
pa2ga0=0xff65,0x195a,0xfd03
AvVmid_c0=0x0,0xc8
cckpwroffset0=5

# PPR params
maxp2ga0=74
txpwrbckof=6
cckbw202gpo=0
legofdmbw202gpo=0x44444444
mcsbw202gpo=0x88888888

# OFDM IIR :
ofdmdigfilttype=18
ofdmdigfilttypebe=18
# PAPD mode:
papdmode=1
papdvalidtest=1

# LTECX flags
ltecxmux=1
ltecxpadnum=0x02030401
ltecxfnsel=0x3003
ltecxgcigpio=0x3012

il0macaddr=00:90:4c:c5:12:38
wl0id=0x431b

deadman_to=0xffffffff
# muxenab: 0x1 for UART enable, 0x10 for Host awake
muxenab=0x10
# CLDO PWM voltage settings - 0x4 - 1.1 volt
#cldo_pwm=0x4
```

chombourger, 2021-12-30:
it would even be better to name that file: `/lib/firmware/brcm/brcmfmac43430-sdio.friendlyarm,nanopi-r1.txt`

chombourger, 2021-12-30:
```
root@mtda:~# iwlist wlan0 scanning
wlan0     Scan completed :
          Cell 01 - Address: 60:E3:27:88:CA:9F
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
                    Quality=24/70  Signal level=-86 dBm  
                    Encryption key:on
                    ESSID:"HOMEAP"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 72ms ago
                    IE: Unknown: 0006484F4D454150
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 03010B
                    IE: Unknown: 2A0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (3) : PSK unknown (6) unknown (8)
                    IE: Unknown: 0B050400140000
                    IE: Unknown: 3B025100
                    IE: Unknown: 2D1AED111BFFFFFF0000000000000000000100000000000000000000
                    IE: Unknown: 3D160B000400000000000000000000000000000000000000
                    IE: Unknown: 7F080400000000000140
                    IE: Unknown: DD180050F2020101810003A4000027A4000042435E0062322F00
```

vj-kumar, 2021-12-30:
/lib/firmware/brcm/brcmfmac43430-sdio.AP6212.txt works(scanning). Just renamed to /lib/firmware/brcm/brcmfmac43430-sdio.txt. That is what the driver expects.

chombourger, 2021-12-30:
> /lib/firmware/brcm/brcmfmac43430-sdio.AP6212.txt works(scanning). Just renamed to /lib/firmware/brcm/brcmfmac43430-sdio.txt. That is what the driver expects.

it will first try the machine specific .txt file and fallback to the generic one if not found
```
[  368.671863] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[  368.682169] brcmfmac mmc0:0001:1: firmware: direct-loading firmware brcm/brcmfmac43430-sdio.bin
[  368.698930] brcmfmac mmc0:0001:1: firmware: direct-loading firmware brcm/brcmfmac43430-sdio.friendlyarm,nanopi-r1.txt
```
(no errors when using brcm/brcmfmac43430-sdio.friendlyarm,nanopi-r1.txt and wlan0 is up)
```
[   20.647324] brcmfmac mmc0:0001:1: firmware: direct-loading firmware brcm/brcmfmac43430-sdio.bin
[   20.663411] brcmfmac mmc0:0001:1: firmware: failed to load brcm/brcmfmac43430-sdio.friendlyarm,nanopi-r1.txt (-2)
[   20.683480] brcmfmac mmc0:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.friendlyarm,nanopi-r1.txt failed with error -2
[   20.706912] brcmfmac mmc0:0001:1: firmware: direct-loading firmware brcm/brcmfmac43430-sdio.txt
```
(error printed but it then loaded brcm/brcmfmac43430-sdio.txt and I therefore got my wlan0 interface up)

vj-kumar, 2021-12-30:
sudo nmcli dev wifi connect  <ssid-name> password <password>

closed 2022-01-02 by chombourger

vj-kumar, 2022-01-02:
FWIW, https://lore.kernel.org/linux-firmware/20220102181025.95291-1-vijaikumar.kanagarajan@gmail.com/T/#u

commit 33d4a0ba (2022-01-02): nanopi-r1: add support for Wi-Fi

Closes: #125