siemens/mtda #128: Hardware support for BeagleBone Black

chombourger, 2021-12-30:


vj-kumar, 2021-12-30:
I also seem to have a BBB rev c :)

chombourger, 2021-12-30:
I created this issue because Hemanth is still using his bbb to run mtda but probably with a Debian 10 base. Since we do not require a lot, we may be able to build a Debian 11 image that uses both mainline u-boot and kernel

hkoturap, 2022-01-03:
Presently we were using https://github.com/MentorEmbedded/industrial-mtda/blob/mel/master/mtda-bbb-for-ipc.yml to build bbb images.Was facing following issue when trying to build                                                       

hemanth@hemanth:/home/hemanth/industrial-mtda/build$ ../industrial-core/scripts/mel-create-stick

# checking build settings...

ERROR: ParseError at /home/hemanth/industrial-mtda/isar/meta/classes/image.bbclass:178: Could not inherit file classes/${PRODUCTION_IMAGE_TYPE}.bbclass

warning: '' is not a supported compression format. Using bzip2 instead

commit 9eaf6660 (2022-01-31): kas/debian: add kas file for the beaglebone-black

Closes: #128
commit c64202f6 (2022-01-31): meta-isar: add beaglebone-black machine

Closes: #128
closed 2022-01-31 by chombourger