siemens/mtda #130: Nanopi R1: See if UART1(external uart) can be used

vj-kumar, 2021-12-30:
Once you put the R1 inside its metallic case, you no longer have access to the debug uart. Only UART1 near the Antenna connector is visible. Need to find out if UART1 can be used as debug uart.

chombourger, 2022-01-04:
It is something we may want as a build-time setting since some folks may want to connect UART1 to their DUT

vj-kumar, 2022-01-04:
Agreed.

bovi, 2023-06-07:
@chombourger do you actually use the NanoPI NEO with a second UART based on the ISAR image in this repo? A colleague from me who is just setting this up is struggling with it. I notice that the documentation seems to elaborate quiet a bit about the NanoPI, hence I would assume it is his fault (-; 

/cc @kergos

chombourger, 2023-06-08:
> @chombourger do you actually use the NanoPI NEO with a second UART based on the ISAR image in this repo? A colleague from me who is just setting this up is struggling with it. I notice that the [...]

Hello @bovi , this issue is for the https://www.friendlyelec.com/index.php?route=product/product&product_id=248, another NanoPI variant.

I don't think the NEO has pins for 2 UARTs: just one.

I did use the UART at the very beginning to bring-up the NEO but have not since (I however believe the first and only UART should still work).

bovi, 2023-06-20:
Hi @chombourger,

the NEO definitly has more than one UART: [image]

As a matter of fact @kergos has managed today to use the UART from a regular Armbian by modifying the overlay kernel configuration. So the hardware is definitely capable but we didn't yet manage to bring this back into ISAR. If we manage to change the ISAR to work probably on the NEO Core would you accept a merge request to integrate it here? Or would you prefer to maintain additional ISAR configurations outside of the tree?

chombourger, 2023-06-21:
thanks for the pointers. I would definitely accept a merge request to get those UART interfaces supported (it would be super useful to have the NanoPI+MTDA able to get the target console over UART)

fmoessbauer, 2023-09-06:
> As a matter of fact @kergos has managed today to use the UART from a regular Armbian by modifying the overlay kernel configuration. So the hardware is definitely capable but we didn't yet manage to bring this back into ISAR.

I'm currently working on integrating the necessary infrastructure for that into ISAR and Linux:

- ISAR: https://groups.google.com/g/isar-users/c/dBuQR-sZRkg (please help review)
- Linux: https://www.spinics.net/lists/devicetree/msg624020.html (accepted, will be part of 6.6 https://github.com/torvalds/linux/commit/0801a3a9f35854c91f1f3ab8e879c3d14912d214 )

The ISAR-only strategy is to use the DTs from u-boot, as these are compile with symbol information. This also works with the distro kernels pre Linux 6.6, however the device trees might be a bit out-of-sync with the one from the corresponding kernel version.  From Linux 6.6 on, we can use the DT from the distro kernel again.

Note: we are already running the ISAR + u-boot DTs approach for some weeks in our lab now.