siemens/mtda #134: mtda_apt_uri: apt update error "Bad Gateway"

ricbha, 2022-01-04:
Error on running apt-get update:
`W: Failed to fetch https://dl.bintray.com/chombourger/mtda/dists/bullseye/InRelease  502  Bad Gateway [IP: 54.70.2.80 443]
`
As added from https://github.com/siemens/mtda/blob/master/meta-isar/recipes-core/images/mtda-image.inc#L81 :
```
mtda@mtda:~$ cat /etc/apt/sources.list.d/mtda.list
deb [trusted=yes] https://dl.bintray.com/chombourger/mtda bullseye main
```

chombourger, 2022-01-04:
right, bintray should be removed since the service was sunset sometime last year

chombourger, 2022-01-04:
looks like we need to update meta-isar/recipes-core/images/mtda-image.inc

chombourger, 2022-01-04:
would you like to propose a fix @ricbha ?

ricbha, 2022-01-04:
> chombourger/mtda

Yes ofcourse. Do we have apt available at gemfury or we need to create one?

chombourger, 2022-01-04:
> > chombourger/mtda
> 
> Yes ofcourse. Do we have apt available at gemfury or we need to create one?

cool. yes we do have feeds up. see https://github.com/siemens/mtda/blob/master/docs/install.rst#using-apt

The entry should now read:
```
deb [trusted=yes] https://apt.fury.io/mtda/ /
```

PR #135 (2022-01-04): apt: mtda apt link changed
Closes: #134
closed 2022-01-04 by chombourger