siemens/mtda #148: storage: mtda-usb-functions service failure

ricbha, 2022-01-10:
mtda-usb-functions service fails mostly on reboot with following error:

```
Jan 04 18:29:27 mtda systemd[1]: Started USB functions for MTDA.
Jan 04 18:29:28 mtda usb-functions[312]: FileNotFoundError: [Errno 2] No such f>
Jan 04 18:29:28 mtda usb-functions[312]: During handling of the above exception>
Jan 04 18:29:28 mtda usb-functions[312]: Traceback (most recent call last):
Jan 04 18:29:28 mtda usb-functions[312]:   File "/usr/libexec/mtda/usb-function>
Jan 04 18:29:28 mtda usb-functions[312]:     install()
Jan 04 18:29:28 mtda usb-functions[312]:   File "/usr/libexec/mtda/usb-function>
Jan 04 18:29:28 mtda usb-functions[312]:     fileputcontent(path + "/functions/>
Jan 04 18:29:28 mtda usb-functions[312]:   File "/usr/libexec/mtda/usb-function>
Jan 04 18:29:28 mtda usb-functions[312]:     fp.write(bytearray(content))
Jan 04 18:29:28 mtda usb-functions[312]: FileNotFoundError: [Errno 2] No such f>
Jan 04 18:29:28 mtda systemd[1]: mtda-usb-functions.service: Main process exite>
Jan 04 18:29:28 mtda systemd[1]: mtda-usb-functions.service: Failed with result>
```

Reason for failure is because this service is looking for /dev/sda even before it is loaded
```
[   15.056464] Mass Storage Function, version: 2009/09/11                       
[   15.061874] LUN: removable file: (no medium)                                 
[   15.064769] mass_storage.usb0/lun.0: unable to open backing file: /dev/sda   
[   15.084285] mass_storage.usb0/lun.0: unable to open backing file: /dev/sda         
[   28.634113] scsi 0:0:0:0: Attached scsi generic sg0 type 0                   
[   28.701273] sd 0:0:0:0: [sda] 60678144 512-byte logical blocks: (31.1 GB/28.)
[   28.710382] sd 0:0:0:0: [sda] Write Protect is off                           
[   28.715332] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00                        
[   28.721297] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doA
[   29.002663]  sda: sda1 sda2                                                  
[   29.012035] sd 0:0:0:0: [sda] Attached SCSI removable disk                   

```

Suggested solution: Modify 10-after-sda.conf to include sleep
Current file
```
mtda@mtda:~$ cat /lib/systemd/system/mtda-usb-functions.service.d/10-after-sda.conf
# Copyright (C) 2021 Siemens Digital Industries Software
# SPDX-License-Identifier: MIT
[Unit]
After=dev-sda.device
```
Suggested file
```
mtda@mtda:~$ cat /lib/systemd/system/mtda-usb-functions.service.d/10-after-sda.conf
# Copyright (C) 2021 Siemens Digital Industries Software
# SPDX-License-Identifier: MIT
[Unit]
After=dev-sda.device

[Service]
ExecStart= /bin/sleep 10

```

chombourger, 2022-01-11:
We probably want to check the solution suggested there: https://lists.fedoraproject.org/pipermail/devel/2012-January/160917.html

chombourger, 2022-01-11:
And there: https://unix.stackexchange.com/questions/186899/how-to-wait-for-a-dev-video0-before-starting-a-service

chombourger, 2022-01-11:
I however see that `/dev/sda` already has the `systemd` tag:
```
udevadm info /dev/sda
sudo: unable to resolve host mtda: No address associated with hostname
[sudo] password for mtda: 
P: /devices/platform/soc/1c1c000.usb/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda
N: sda
L: 0
S: disk/by-id/usb-SanDisk_Ultra_USB_3.0_4C530000150318215551-0:0
S: disk/by-path/platform-1c1c000.usb-usb-0:1:1.0-scsi-0:0:0:0
E: DEVPATH=/devices/platform/soc/1c1c000.usb/usb4/4-1/4-1:1.0/host0/target0:0:0/0:0:0:0/block/sda
E: DEVNAME=/dev/sda
E: DEVTYPE=disk
E: MAJOR=8
E: MINOR=0
E: SUBSYSTEM=block
E: USEC_INITIALIZED=12795699
E: ID_VENDOR=SanDisk
E: ID_VENDOR_ENC=SanDisk\x20
E: ID_VENDOR_ID=0781
E: ID_MODEL=Ultra_USB_3.0
E: ID_MODEL_ENC=Ultra\x20USB\x203.0\x20\x20\x20
E: ID_MODEL_ID=5591
E: ID_REVISION=1.00
E: ID_SERIAL=SanDisk_Ultra_USB_3.0_4C530000150318215551-0:0
E: ID_SERIAL_SHORT=4C530000150318215551
E: ID_TYPE=disk
E: ID_INSTANCE=0:0
E: ID_BUS=usb
E: ID_USB_INTERFACES=:080650:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=usb-storage
E: ID_PATH=platform-1c1c000.usb-usb-0:1:1.0-scsi-0:0:0:0
E: ID_PATH_TAG=platform-1c1c000_usb-usb-0_1_1_0-scsi-0_0_0_0
E: ID_PART_TABLE_UUID=d97178b3-0416-40ea-b3db-34d860efd77e
E: ID_PART_TABLE_TYPE=gpt
E: DEVLINKS=/dev/disk/by-id/usb-SanDisk_Ultra_USB_3.0_4C530000150318215551-0:0 /dev/disk/by-path/platform-1c1c000.usb-usb-0:1:1.0-scsi-0:0:0:0
E: TAGS=:systemd:
E: CURRENT_TAGS=:systemd:
```

ricbha, 2022-01-11:
@ptsarath tried mtda-14, faced same issue. For me adding "Wants=/dev/sda"  before "After=/dev/sda" solved the issue.

ricbha, 2022-01-11:
> @ptsarath tried mtda-14, faced same issue. For me adding "Wants=/dev/sda" before "After=/dev/sda" solved the issue.

@chombourger

chombourger, 2022-01-11:
Good news,let's go with this change!

ricbha, 2022-01-11:
@chombourger should doc/fixtures.rst be updates as well?

PR #150 (2022-01-12): storage: start mtda-usb-functions after storage is loaded
Closes: #148
closed 2022-01-12 by chombourger