siemens/mtda #151: keyboard/hid: broken pipe

chombourger, 2022-01-11:
Gunther experienced the following issue with one his IPCs
[image]

chombourger, 2022-01-11:
Here's the power script from his setup:
```
[scripts]
power on:
	variant = 'ipc227e'
	if 'variant' in env:
	... variant = env['variant']
	... mtda.debug(1, "!!!Variant is set")
	if 'boot-from-usb' in env and env['boot-from-usb'] == '1':
	... import time
	... time.sleep(5)
	... mtda.keyboard.probe()
	... mtda.debug(1, "!!!Keyboard is probed")
	... mtda.keyboard.esc(5)
	... mtda.debug(1, "!!!Sent ESC to DUT")
	... time.sleep(1)
	... if variant == 'ipc227e':
	... ... mtda.keyboard.right()
	... ... mtda.debug(1, "!!!Variant is set as ipc227e")
	... elif variant == 'ipc127e':
	... ... mtda.keyboard.down()
	... ... mtda.debug(1, "!!!Variant is set as ipc127e")
	... mtda.keyboard.enter()
	... time.sleep(0.5)
	... mtda.keyboard.down()
	... mtda.keyboard.enter()
	... mtda.keyboard.idle()
```
he then said that using sleep(5) or sleep(6) does not work but sleep(5.5) does

chombourger, 2022-01-11:
I had experienced a similar issue with OCII platforms because the BIOS would not support the USB composite device we provide and would not respond to the USB reports we send

hkoturap, 2022-01-19:
Presently with couple of QA Setups we are experiencing the issue of `BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown` after switching to 0.14 MTDA  version.
```
# debug3: keyboard.hid.write_report()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 87, in press
    written = self.write_report(
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 56, in write_report
    return self.fd.write(report.encode())
BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
```

Below is the complete log.
```
mtda@mtda:~$ sudo mtda-cli target on
# debug3: main.load_environment()
# debug4: main.load_environment(): variant => ipc427e
# debug3: env_set()
# debug3: main._session_check(None)
# debug3: main._session_check: None
# debug3: env_set(): None
# debug3: main.load_timeouts_config()
# debug3: main.load_timeouts_config: None
# debug3: main.load_power_config()
# debug3: main.load_console_config()
# debug3: console.serial.configure()
# debug3: main.load_keyboard_config()
# debug3: keyboard.hid.configure()
# debug4: keyboard.hid.configure(): will use /dev/hidg0
# debug3: keyboard.hid.probe()
# debug3: keyboard.hid.probe(): True
# debug3: main.load_storage_config()
# debug3: storage.writer.compression.set()
# debug3: storage.writer.compression.set(): IMAGE.RAW
# debug3: storage.usbf.configure()
# debug3: storage.usbf.configure(): None
# debug3: main._parse_script()
# debug3: main._parse_script(): 
# debug3:  ... from time import sleep
# debug3:  ... sleep(5)
# debug3:  ... mtda.keyboard.esc(5)
# debug3:  ... sleep(1)
# debug3:  ... mtda.keyboard.right()
# debug3:  ... sleep(1)
# debug3:  ... mtda.keyboard.enter()
# debug3:  ... sleep(1)
# debug3:  ... mtda.keyboard.down()
# debug3:  ... mtda.keyboard.down()
# debug3:  ... mtda.keyboard.down()
# debug3:  ... mtda.keyboard.enter()
# debug3:  ... mtda.keyboard.idle()
# debug3: main._parse_script()
# debug3: main._parse_script(): None
# debug3: main.start()
# debug3: storage.usbf.probe()
# debug3: storage.usbf.probe(): True
# debug3: console.serial.probe()
# debug3: console.serial.probe(): True
# debug3: main.target_on()
# debug3: console.serial.open()
# debug3: console.serial.open(): True
# debug3: main._session_check(unsparing)
# debug2: session unsparing: 300 seconds
# debug3: main._session_event(ACTIVE unsparing)
# debug3: main.notify()
# debug3: main.notify: None
# debug3: main._session_event: None
# debug3: main._session_check: None
# debug3: main.power_locked()
# debug3: main._session_check(unsparing)
# debug2: session unsparing: 300 seconds
# debug3: main._session_check: None
# debug3: main._check_locked()
# debug3: main.target_owner()
# debug3: main.power_locked(): False
Command sent successfully.
# debug3: main.exec_power_on_script()
# debug3: # debug4: console.serial.open()
# debug4: console.serial.open(): already opened
# debug3: console.serial.open(): True
exec_power_on_script(): 
# debug4:  ... from time import sleep
# debug3: # debug4:  ... sleep(5)
console.serial.pending()
# debug4:  ... mtda.keyboard.esc(5)
# debug3: # debug4:  ... sleep(1)
console.serial.pending(): 0
# debug4:  ... mtda.keyboard.right()
# debug3: # debug4:  ... sleep(1)
console.serial.read()
# debug4:  ... mtda.keyboard.enter()
# debug4:  ... sleep(1)
# debug4:  ... mtda.keyboard.down()
# debug4:  ... mtda.keyboard.down()
# debug4:  ... mtda.keyboard.down()
# debug4:  ... mtda.keyboard.enter()
# debug4:  ... mtda.keyboard.idle()
# debug3: keyboard.hid.esc()
# debug3: keyboard.hid.press()
# debug4: keyboard.hid.press(): opening /dev/hidg0
# debug3: keyboard.hid.write_report()
# debug3: keyboard.hid.write_report()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 87, in press
    written = self.write_report(
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 56, in write_report
    return self.fd.write(report.encode())
BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/mtda-cli", line 816, in <module>
    app.main()
  File "/usr/bin/mtda-cli", line 802, in main
    status = cmds[cmd](stuff)
  File "/usr/bin/mtda-cli", line 688, in target_cmd
    return cmds[cmd](args)
  File "/usr/bin/mtda-cli", line 649, in target_on
    status = self.client().target_on()
  File "/usr/lib/python3/dist-packages/mtda/client.py", line 304, in target_on
    return self._impl.target_on(self._session)
  File "/usr/lib/python3/dist-packages/mtda/main.py", line 764, in target_on
    self.exec_power_on_script()
  File "/usr/lib/python3/dist-packages/mtda/main.py", line 748, in exec_power_on_script
    result = exec(self.power_on_script,
  File "<string>", line 4, in <module>
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 110, in esc
    return self.press(0x29, 0, repeat)
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 97, in press
    self.write_report(NULL_CHAR * 8)
  File "/usr/lib/python3/dist-packages/mtda/keyboard/hid.py", line 56, in write_report
    return self.fd.write(report.encode())
BrokenPipeError: [Errno 108] Cannot send after transport endpoint shutdown
resetting console to recover from read error (<class 'serial.serialutil.SerialException'>)...
# debug3: console.serial.close()
# debug3: console.serial.close(): True
# debug3: console.serial.open()
# debug3: console.serial.open(): True
```

chombourger, 2022-01-19:
just tried with packages built from HEAD on top of a NanoPI image that was still on a 4.19 kernel and keyboard events are sent without errors/timeouts

hkoturap, 2022-01-20:
After updating NanoPI with image having 4.19 kernel we don't observe any error with keyboard events.

PR #173 (2022-01-21): keyboard/hid: make write_report() non-blocking
Give-up after a second if we cannot write the HID report.

Fixes: #151
commit ae7d9e30 (2022-01-21): keyboard/hid: make write_report() non-blocking

Give-up after a second if we cannot write the HID report.

Fixes: #151
closed 2022-01-21 by chombourger