siemens/mtda #156: mtda service fails after installation

bhargavdas, 2022-01-13:
While installing mtda package following instruction from https://github.com/siemens/mtda/blob/master/docs/install.rst#using-apt , in a fresh installation of ubuntu 20.04 or debian 11
it is observed the service fails after installation.

Here is the log
```
bhargav@ubuntu:~$ echo 'deb [trusted=yes] https://apt.fury.io/mtda/ /' | \
  sudo tee /etc/apt/sources.list.d/mtda.list
[sudo] password for bhargav: 
deb [trusted=yes] https://apt.fury.io/mtda/ /
bhargav@ubuntu:~$ sudo apt-get update
Hit:1 http://us.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://security.ubuntu.com/ubuntu jammy-security InRelease                         
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease                        
Hit:4 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease
Ign:5 https://apt.fury.io/mtda  InRelease     
Ign:6 https://apt.fury.io/mtda  Release
...
...
Ign:9 https://apt.fury.io/mtda  Translation-en_US
Fetched 17.5 kB in 5s (3,457 B/s)
Reading package lists... Done
bhargav@ubuntu:~$ sudo apt-get install -y mtda
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libc-ares2 libev4 libhidapi-hidraw0 libnorm1 libpgm-5.3-0 libzmq5 python3-daemon python3-future python3-gevent python3-greenlet python3-lib2to3 python3-lockfile python3-msgpack python3-psutil
  python3-serial python3-usb python3-zerorpc python3-zmq python3-zope.event python3-zope.interface python3-zstandard usbrelay
Suggested packages:
  python-future-doc python-gevent-doc python3-openssl python-greenlet-dev python-greenlet-doc python-lockfile-doc python-psutil-doc python3-wxgtk3.0 | python3-wxgtk python-zstandard-doc
The following NEW packages will be installed:
  libc-ares2 libev4 libhidapi-hidraw0 libnorm1 libpgm-5.3-0 libzmq5 mtda python3-daemon python3-future python3-gevent python3-greenlet python3-lib2to3 python3-lockfile python3-msgpack
  python3-psutil python3-serial python3-usb python3-zerorpc python3-zmq python3-zope.event python3-zope.interface python3-zstandard usbrelay
0 upgraded, 23 newly installed, 0 to remove and 39 not upgraded.
Need to get 3,702 kB of archives.
After this operation, 16.9 MB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 libnorm1 amd64 1.5.9+dfsg-2 [221 kB]
Get:3 https://apt.fury.io/mtda  python3-zerorpc 0.6.3-2 [19.9 kB]       
...
...
bhargav@ubuntu:~$ sudo journalctl -u mtd
-- Journal begins at Thu 2022-01-13 11:38:22 PST, ends at Thu 2022-01-13 11:49:18 PST. --
-- No entries --
bhargav@ubuntu:~$ sudo journalctl -u mtda
-- Journal begins at Thu 2022-01-13 11:38:22 PST, ends at Thu 2022-01-13 11:49:20 PST. --
Jan 13 11:48:42 ubuntu systemd[1]: Started Multi-Tenant Device Access.
Jan 13 11:48:42 ubuntu systemd[1]: Stopping Multi-Tenant Device Access...
Jan 13 11:48:42 ubuntu systemd[1]: mtda.service: Deactivated successfully.
Jan 13 11:48:42 ubuntu systemd[1]: Stopped Multi-Tenant Device Access.
Jan 13 11:48:42 ubuntu systemd[1]: Started Multi-Tenant Device Access.
Jan 13 11:48:42 ubuntu mtda-cli[3780]: Traceback (most recent call last):
Jan 13 11:48:42 ubuntu mtda-cli[3780]:   File "/usr/bin/mtda-cli", line 26, in <module>
Jan 13 11:48:42 ubuntu mtda-cli[3780]:     import zeroconf
Jan 13 11:48:42 ubuntu mtda-cli[3780]: ModuleNotFoundError: No module named 'zeroconf'
Jan 13 11:48:42 ubuntu systemd[1]: mtda.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 11:48:42 ubuntu systemd[1]: mtda.service: Failed with result 'exit-code'.
Jan 13 11:48:42 ubuntu systemd[1]: Started Multi-Tenant Device Access.
Jan 13 11:48:42 ubuntu systemd[1]: Stopping Multi-Tenant Device Access...
Jan 13 11:48:42 ubuntu systemd[1]: mtda.service: Deactivated successfully.
Jan 13 11:48:42 ubuntu systemd[1]: Stopped Multi-Tenant Device Access.
Jan 13 11:48:42 ubuntu systemd[1]: Started Multi-Tenant Device Access.
Jan 13 11:48:43 ubuntu mtda-cli[3818]: Traceback (most recent call last):
Jan 13 11:48:43 ubuntu mtda-cli[3818]:   File "/usr/bin/mtda-cli", line 26, in <module>
Jan 13 11:48:43 ubuntu mtda-cli[3818]:     import zeroconf
Jan 13 11:48:43 ubuntu mtda-cli[3818]: ModuleNotFoundError: No module named 'zeroconf'
Jan 13 11:48:43 ubuntu systemd[1]: mtda.service: Main process exited, code=exited, status=1/FAILURE
Jan 13 11:48:43 ubuntu systemd[1]: mtda.service: Failed with result 'exit-code'.
```

chombourger, 2022-01-14:
See https://github.com/siemens/mtda/pull/157#issuecomment-1012931212. It is already fixed, just not released. We will try to get 0.15 out ASAP. The work-around is to `apt-get install python3-zeroconf`

chombourger, 2022-01-14:
Assigning to myself. The fix is to release 0.15 :)

chombourger, 2022-01-14:
Same as #132

closed 2022-01-20 by chombourger