siemens/mtda #159: build broken with latest Isar HEAD

chombourger, 2022-01-14:
The following commits were recently added to Isar's master:
```
74036eb vm-img: set layer-based template search path
d158150 image: Protect shared deploy directory usage
16ed6c3 image: Make mount points imagetype-specific
9724e3b Revert "buildchroot: Do not build cross when there are no arch-specific outputs"
eeefa03 Revert "buildhcroot/build.sh: Disable deps checking"
6e57258 debian-common: Add fdisk package as a dependency in wic
1a17e37 meta-isar: switch virtualbox test from buster to bullseye
6dc7dec vm-img: make use of isars template class for most of the tmpl variables
4103d79 vm-img: set vmware machine type to 64bit in our template
5f55687 vm-img: drop not used variables
40ebc41 vm-img: shorten code by using variables we have
02f9616 vm-img: use two different ovf templates for vmware/virtualbox
f7f431a vm-img: move template out of package into SRC_URI file
05aed40 Adjust debian-bullseye.list to Debian recommendations
d00f70c sstate: Make image cache imagetype-specific
a0e4e19 classes: fix comment in container-img class
4a18e8e classes: rename CONTAINER_FORMATS to CONTAINER_IMAGE_FORMATS
18f6e63 classes: allow more control over container image name and tag
a21e26b classes: simplify tag handling in container class
08804af classes: make sure container extension can run multiple times
19c3126 isar-exclude-docs: include "native" package changelogs
908fc86 debianize: prepend changelog entries
53d315f debianize: make changelog date reproducible
c32de06 isar-bootstrap: Remove unused line
cbd5a3a isar-bootstrap: Fail on invalid apt source list
be1f692 testsuite: drop wic exclude path test
165519a sudo: Fail on the first error
cf25bf2 rootfs: Execute rmdir only if needed
d38ed72 refactor: inject ccache environment similar to proxy env
c408740 Setup ccache in devshell as well
da90ff8 conf/distro: Fix deb-src for debian-sid-ports
b524d64 conf: Use debian-bullseye by default
837b58e ci: Add hikey target
d447f99 meta-isar: Enable bullseye support for HiKey board
9d12af3 sstate: add caching to debian packages
085efc7 sstate: add caching to rootfs
e116e48 sstate: add caching to isar-bootstrap
0a18d4c sstate: configure
82aa2ee meta: add sstate feature from oe
c32eddf dpkg: add explicit dependency to isar-apt
dca1471 base: remove unneeded "before do_build" task dependencies
7c7628e rootfs: recursively depend on packages
800a6fe images: create deploy dir
40bb2fd doc: Add buildstats section in user manual
2e64903 buildstats: Enable buildstats in default configuration
dbfbfee buildstats: Fixes to work with bbclass in ISAR
c1c0807 buildstats: Borrow buildstats and pybootchartgui from OE
74f23a5 oe imports in central location
852a9b0 api: Rename IMAGE_TYPE to IMAGE_FSTYPES
2f4065c doc: Update docs to fit new IMAGE_TYPE meaning
5c59cde multiconfig: Add ext4-img type to qemuamd64-buster multiconfig.
4120053 start_vm: Use the first image type to start VM
f0068f6 wic-img: Set weak default value for WKS_FILE
a9c9e36 image: Support multiple image types in WORKDIR and STAMPS
```
Our builds are now broken:
```
ERROR: /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb: Invalid apt sources list
ERROR: ExpansionError during parsing /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb
```

jan-kiszka, 2022-01-14:
sstate-related, likely.

Why are you taking this strange indirection via `MTDA_EXTRA_APT_SOURCES`, rather than appending the target variables directly in the machine confs? Not saying that this is the reason, but it looks like a pointless abstraction.

chombourger, 2022-01-14:
> sstate-related, likely.
> 
> Why are you taking this strange indirection via `MTDA_EXTRA_APT_SOURCES`, rather than appending the target variables directly in the machine confs? Not saying that this is the reason, but it looks like a pointless abstraction.

you are right. that approach would also cause problems with multiconfig builds (isar-bootstrap-target should be consistent across all machines included in a multiconfig build)

raised #160 as an attempt to clean this up

it builds OK with the older Isar. I am still left with errors with Isar's HEAD:
```
ERROR: /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb: Invalid apt sources list
WARNING: /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb: Exception during build_dependencies for do_bootstrap
WARNING: /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb: Error during finalise of /work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-host.bb
``

these are only observed using kas/debian/nanopi-all.yml (multiconfig)
both kas/debian/nanopi-neo.ym and kas/debian/nanopi-r1.yml are building ok

chombourger, 2022-01-14:
setting `distro` in `kas/debian/mtda-nanopi-all.yml` appears to fix the issue

chombourger, 2022-01-14:
the build goes further now but eventually fails on some Python recipes. investigating

chombourger, 2022-01-15:
> the build goes further now but eventually fails on some Python recipes. investigating

several python packages fail in `do_dpkg_build` complaining that python3-setuptools:native isn't installed while it is a Build-Depends in debian/control and was installed at some point in the buildchroot. Was it removed from the shared buildchroot when installing deps for another package?

jan-kiszka, 2022-01-15:
If some other recipe erroneously requests `python3-setuptools`, the target arch version may get installed into the buildchroot. And if that happens between `do_install_deps` and `do_dpkg_build` of the affected recipe, you have your error. In fact, that may even come from broken upstream packages if you rebuild any (had to fix systemd recently).

commit 3d4b01f8 (2022-01-17): kas/debian: set distro in multiconfig kas files

Fixes: #159
closed 2022-01-17 by chombourger