siemens/mtda #161: support multiple hard disks for qemu

GuBi34, 2022-01-17:
For some use case a virtualized target with multiple hard disks (QEMU-KVM). To support it out of the box the `config.ini` as well as the power/qemu.py scripts needs to be extended.

I would propose to define the storage variable as an array with the following mapping to the `config.ini`

```
storage   -> storage ### for compatibility reasons
storage.0 -> storage[0]
storage.1 -> storage[1]
storage.2 -> storage[2]
```

For compatibility reasons a mapping of `storage -> storage[0]` may be considered.

Default names on disk then would result to
```
/var/lib/mtda/ssd.img ### for compatibility reasons
/var/lib/mtda/ssd0.img
/var/lib/mtda/ssd1.img
....
```

chombourger, 2022-01-17:
so we would want both ssd0 and ssd1 to be drives we could swap between the host and target (DUT)?
note that the kvm configuration already comes with 3 drives:

- ssd.img: fixed drive (cannot be seen from host using MTDA just like the hard-drive on real hardware)
- usb-shared-storage.img: the drive that may be swapped between the host and target (DUT) using the storage command
- usb-data-storage.img: the drive that may be plugged in/out using the usb command

GuBi34, 2022-01-17:
In our case the intended execution environment has two physical (virtual) drives for system and userdata. Both need not to be swaped. They belong to the system as hard drives. So indeed I mean ssdX.img (not to be used/seen via MTDA).

My proposal is to provide an extended configuration possibilities regarding the hard drives to the MTDA user simulation an execution environment with two (or more) hard disks in QEMU.

A prototype is running on it and I can propose changes once we decided that this is desirable for a broader audience.

chombourger, 2022-01-17:
thanks for clarifying the use-case. all clear now and making sense.

chombourger, 2022-01-17:
targeting this issue for 0.15 (to be released end of January)

chombourger, 2022-01-17:
raised #163 - feel free to provide comments there

commit 328b98f6 (2022-01-18): power/qemu: support additional system drives

Support emulation of multiple system drives with "storage.<N>"
configuration entries. If both "storage" and "storage.0" are
set, "storage.0" will be used.

Fixes: #161
PR #163 (2022-01-18): power/qemu: support additional system drives
Support emulation of multiple system drives with "storage.<N>"
configuration entries. If both "storage" and "storage.0" are
set, "storage.0" will be used.

Fixes: #161
closed 2022-01-18 by chombourger