siemens/mtda #168: make mtda-usb-functions expose a second UART port

chombourger, 2022-01-20:
a monitor application could be implemented on the DUT side and made available through the mtda monitor interface

chombourger, 2022-01-21:
implemented but unable to add a second acm
same issue as https://www.spinics.net/lists/linux-usb/msg163414.html

FYI @GuBi34

chombourger, 2022-01-21:
Raised https://github.com/siemens/mtda/pull/177
The kernel on the NanoPI is however failing to create a second ACM instance
So for now we shouldn't add a `monitor` interface with `variant=usbf`

chombourger, 2022-01-21:
Here I got:
```
[24202.126294] configfs-gadget gadget: acm/59ec4bc5: can't bind, err -19
[24202.133838] configfs-gadget musb-hdrc.2.auto: failed to start usb_functions_for_multi-tenant_device_access: -19
```

chombourger, 2022-01-23:
Allwinner OTG supports up to 5 endpoints (the number of functions we are trying to create exceeds the number of buffers (endpoints) supported by the hardware)

mediatek appears to support up to 8

GuBi34, 2022-01-24:
Thank you for exploration. I question if we cannot utilize the monitor we already have (for qemu under `mtda-cli command ...`) for this purpose (or am I missunderstanding things here)?

GuBi34, 2022-01-24:
Thanks for the exploration.

I wonder if we could/should mux e.g. serial console with monitor (just thinking out loud).

chombourger, 2022-01-30:
works on the BeagleBone Black (hardware support proposed with #181)

chombourger, 2022-01-30:
sample config for the bbb
```
[main]
debug=2

[environment]
variant=ipc227e

[console]
variant=usbf

[monitor]
variant=usbf

[keyboard]
variant=hid
device=/dev/hidg0

[storage]
variant=usbf
file=/dev/mmcblk0

[power]
variant=gpio
pin=48

[usb]
ports=1

[usb1]
class=MSC
variant=gpio
pin=20
enable=high

[scripts]
power on:
    variant = 'ipc227e'
    if 'variant' in env:
    ... variant = env['variant']
    if 'boot-from-usb' in env and env['boot-from-usb'] == '1':
    ... import time
    ... time.sleep(5)
    ... mtda.keyboard.esc(5)
    ... time.sleep(1)
    ... if variant == 'ipc227e':
    ... ... mtda.keyboard.right()
    ... elif variant == 'ipc127e':
    ... ... mtda.keyboard.down()
    ... mtda.keyboard.enter()
    ... time.sleep(0.5)
    ... mtda.keyboard.down()
    ... mtda.keyboard.enter()
    ... mtda.keyboard.idle()
```

closed 2022-01-31 by chombourger