siemens/mtda #185: Config generator for mtda configs

vj-kumar, 2022-02-03:
It would be nice to have some kind of config generator. I see that some users are not aware of the latest config options. It would help them to know about the option and any changes in the default values.

chombourger, 2022-02-03:
That and maybe a `fix` command like found in packer (https://www.packer.io/docs/commands/fix)

vj-kumar, 2022-02-03:
Yes. That would be much helpful.

vj-kumar, 2022-02-26:
@chombourger Started investigating this.

chombourger, 2022-02-28:
> @chombourger Started investigating this.

Cool. thank you!

vj-kumar, 2022-03-28:
This is what the initial PoC looks like.
Used Kconfiglib with Urwid.
[image]

vj-kumar, 2022-03-28:
Still some more work before it can be ready.
1. Group the configs based on sections in UI.
2. Establish dependencies
3. Add feature to load existing ini files.
4. Add support to select a deselect a section(Related to 2)