siemens/mtda #213: the mac address on the nanopi-r1 changes from boot to boot

GuBi34, 2022-02-14:
The nanopi-r1 provides every boot a different MAC address.

Reproduction:
Build mtda for nanopi-r1
Flash the card
Boot
Login -> readout ip addr
Power off/on
Login -> readout ip addr

chombourger, 2022-02-14:
and so does the NEO

GuBi34, 2022-02-14:
On my nanopi-neo tha MAC address appears to be unchanged after power cycling
It is alway registered in the known_hosts and receives the same IP address on my hom router (not so for the nanopi-r1)
so there seem to be some differences.
Obviously the changing MAC causes difficulties at least if the setup is  not configured for fixed IP on the device.

chombourger, 2022-02-14:
you could do this:
```
nmcli con modify 'Wired connection 1' 802-3-ethernet.cloned-mac-address 00:12:34:56:78:9a
```
for NetworkManager to use a custom MAC address. We can add this to the FAQ
We could also implement a systemd service to somehow derive a MAC address from the serial number of the CPU

GuBi34, 2022-02-15:
Probably the the auto generation of a random MAC provides sufficient added value to be considered?! - anyhow I try to follow your advice to change the connection manually besides of this!

chombourger, 2022-02-15:
> Probably the the auto generation of a random MAC provides sufficient added value to be considered?! - anyhow I try to follow your advice to change the connection manually besides of this!

indeed. I have started to work on a systemd service to derive a MAC address from the serial number of the CPU

chombourger, 2022-02-22:
note: not finding to work on this right now :(