siemens/mtda #214: support disabling the mass storage function at runtime

chombourger, 2022-02-16:
Some devices do not allow the boot media to be configured and will instead check if a (valid) USB stick is connected before booting the system. We may need a disable() function to get the mass storage function removed from the USBF stack

badrikeshp-mg, 2022-03-03:
Hi @chombourger , is this issue related to:
1. That is currently faced on older comfort boards, where the shared storage gets detected on the target even when it attached to host. And the disable function will help so that shared storage won't get detected on target side
2. Or any new enhancement for new comfort hmi boards that can help in bypassing maintenance key press to boot to service stick

chombourger, 2022-03-03:
> Hi @chombourger , is this issue related to:
>
> 1. That is currently faced on older comfort boards, where the shared storage gets detected on the target even when it attached to host. And the disable [...]

This is about (1)

commit fd71061d (2022-12-01): support/usb: enable/disable USB functions on power events

Re-create USB functions that are needed from target_on() and
uninstall all of them from target_off().

Closes: #214
closed 2022-12-01 by chombourger