siemens/mtda #219: Mtda image (mtda-nanopi-neo.yml) mtda-cli won't work on NanoPI NEO-LTS

Urban-Kosale, 2022-02-28:
Mtda image (mtda-nanopi-neo.yml) mtda-cli won't work on NanoPI NEO-LTS

After a successful build of [mtda-nanopi-neo.yml] image for NanoPI NEO-LTS with instructions from [https://mtda.readthedocs.io/en/latest/build.html](url) we tested the relay function (mtda-cli target on; mtda-cli target of) with same configuration as it is described in mtda.readthedocs.io and came to a conclusion that something doesn't work out of the box. If we try to control relay with echo "1" > /sys/class/gpio/gpio203/value relay works. But if we try to control relay with mtda-cli it doesn't work. 

Configuration file: 

created with: # vi /etc/mtda/config

content:

_[main]
debug=5

[power]
variant=gpio
pins=203_

If we write: root@mtda:~# mtda-cli target on
we get result:

`root@mtda:~# mtda-cli target on
# debug3: main.load_timeouts_config()
# debug3: main.load_timeouts_config: None
# debug3: main.start()
# debug3: main._target_off()
# debug3: main._target_off(): True
# debug3: main.storage_to_target()
# debug3: main._session_check(None)
# debug3: main._session_check: None
# debug3: main.storage_locked()
# debug3: main._session_check(None)
# debug3: main._session_check: None
# debug3: main._check_locked()
# debug3: main.target_owner()
# debug4: storage_locked(): no shared storage device
# debug3: main.storage_locked(): True
# debug1: storage_to_target(): shared storage is locked
# debug3: main.storage_to_target(): False
# debug3: main.target_on()
# debug3: main._session_check(intervened)
# debug3: session intervened: 300 seconds
# debug3: main._session_event(ACTIVE intervened)
# debug3: main.notify(SESSION ACTIVE intervened)
# debug3: main.notify: None
# debug3: main._session_event: None
# debug3: main._session_check: None
# debug3: main.target_status()
# debug3: main.target_status(): ???
# debug3: main.power_locked()
# debug3: main._session_check(intervened)
# debug3: session intervened: 300 seconds
# debug3: main._session_check: None
# debug3: main.power_locked(): True
# debug3: main.target_on(): False
root@mtda:~# echo $?
1
root@mtda:~# 
`
same is true for mtda-cli target off

systemctl status mtda returns:

`● mtda.service - Multi-Tenant Device Access
     Loaded: loaded (/lib/systemd/system/mtda.service; enabled; vendor preset: >
    Drop-In: /usr/lib/systemd/system/mtda.service.d
             └─10-after-sda.conf
     Active: active (running) since Mon 2022-02-28 12:10:06 UTC; 6s ago
   Main PID: 440 (mtda-cli)
      Tasks: 6 (limit: 1071)
     Memory: 59.3M
        CPU: 6.248s
     CGroup: /system.slice/mtda.service
             └─440 /usr/bin/python3 /usr/bin/mtda-cli -d -n

Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main._session_check: None
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main.storage_locked()
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main._session_check(None)
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main._session_check: None
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main._check_locked()
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main.target_owner()
Feb 28 12:10:12 mtda mtda-cli[440]: # debug4: storage_locked(): no shared stora>
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main.storage_locked(): True
Feb 28 12:10:12 mtda mtda-cli[440]: # debug1: storage_to_target(): shared stora>
Feb 28 12:10:12 mtda mtda-cli[440]: # debug3: main.storage_to_target(): False
`

chombourger, 2022-02-28:
thanks for reaching out and sorry to hear that it isn't working. One thing I noticed in your configuration is a trailing `_` after the pin number:
```
pins=203_
```
is the underscore really there on your side (or is it a copy & paste issue with GitHub)?

chombourger, 2022-02-28:
Note that we should (at the very least) make our configuration parser more robust and "complain" when non-integers values are found in the user-provided config

UKosale, 2022-02-28:
Hi,
thank you for quick response.
it was error copy & paste issue with GitHub.

chombourger, 2022-02-28:
would you be able to provide the output of `journalctl -u mtda` so we see what is going-on on the service side (the logs you provided are for the client-side)

chombourger, 2022-02-28:
and if you could also confirm the version you are running (or git commit hash)

UKosale, 2022-02-28:
mtda@mtda:~$ sudo journalctl -u mtda                                                                                  
-- Journal begins at Tue 2021-07-13 17:29:26 UTC, ends at Mon 2022-02-28 14:09:>
Feb 28 13:54:27 mtda systemd[1]: Started Multi-Tenant Device Access.
Feb 28 14:05:42 mtda systemd[1]: Stopping Multi-Tenant Device Access...
Feb 28 14:05:42 mtda systemd[1]: mtda.service: Succeeded.
Feb 28 14:05:42 mtda systemd[1]: Stopped Multi-Tenant Device Access.
Feb 28 14:05:43 mtda systemd[1]: mtda.service: Consumed 8.082s CPU time.
Feb 28 14:05:43 mtda systemd[1]: Started Multi-Tenant Device Access.
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.load_timeouts_config()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.load_timeouts_config: None
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.load_power_config()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.start()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._target_off()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: power.gpio.status: pin #203 is OFF
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._target_off(): True
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.storage_to_target()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._session_check(None)
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._session_check: None
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.storage_locked()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._session_check(None)
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._session_check: None
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main._check_locked()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.target_owner()
Feb 28 14:05:49 mtda mtda-cli[735]: # debug4: storage_locked(): no shared stora>
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.storage_locked(): True
Feb 28 14:05:49 mtda mtda-cli[735]: # debug1: storage_to_target(): shared stora>
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: main.storage_to_target(): False
Feb 28 14:05:59 mtda mtda-cli[735]: # debug3: main._session_check(None)
Feb 28 14:05:59 mtda mtda-cli[735]: # debug3: main._session_check: None
Feb 28 14:06:09 mtda mtda-cli[735]: # debug3: main._session_check(None)
Feb 28 14:06:09 mtda mtda-cli[735]: # debug3: main._session_check: None

Commit: commit c5f2a7c46d16e0e2b0139d65ca3274ea70ed3086 (HEAD -> master, tag: v0.16-rc1)

chombourger, 2022-02-28:
thank you so much. Logs look good. We even see that it has properly queried GPIO 203
```
Feb 28 14:05:49 mtda mtda-cli[735]: # debug3: power.gpio.status: pin https://github.com/siemens/mtda/discussions/203 is OFF
```
will build this particular commit and check

chombourger, 2022-02-28:
Downloaded our CI build for https://github.com/siemens/mtda/commit/c5f2a7c46d16e0e2b0139d65ca3274ea70ed3086 from https://github.com/siemens/mtda/actions/runs/1865231424

I then created `/etc/mtda/config` as specified in the issue description

The logs I get after starting mtda are as shown in https://github.com/siemens/mtda/issues/219#issuecomment-1054298338

I then ran the following command on my PC:
```
$ mtda-cli -r 192.168.1.44 target on
```
and observed the following logs on the NanoPI (running `journalctl -f -u mtda`):
```
Feb 28 16:33:57 mtda mtda-cli[482]: # debug3: main._session_check(None)
Feb 28 16:33:57 mtda mtda-cli[482]: # debug3: session maltreats: 297 seconds
Feb 28 16:33:57 mtda mtda-cli[482]: # debug3: main._session_check: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.target_on()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_check(kits)
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: session maltreats: 292 seconds
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: session kits: 300 seconds
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_event(ACTIVE kits)
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.notify(SESSION ACTIVE kits)
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.notify: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_event: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_check: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.target_status()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: power.gpio.status: pin #203 is OFF
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.target_status(): OFF
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.power_locked()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_check(kits)
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: session maltreats: 292 seconds
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: session kits: 299 seconds
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_check: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._check_locked()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.target_owner()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.power_locked(): False
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._target_on()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.power_locked()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_check(kits)
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: session maltreats: 292 seconds
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: session kits: 299 seconds
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._session_check: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._check_locked()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.target_owner()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.power_locked(): False
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: power.gpio.status: pin #203 is ON
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.exec_power_on_script()
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.exec_power_on_script(): None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.notify(POWER ON)
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.notify: None
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main._target_on(): True
Feb 28 16:34:01 mtda mtda-cli[482]: # debug3: main.target_on(): True
```
I also checked sysfs:
```
root@mtda:/etc/mtda# cd /sys/class/gpio/gpio203
root@mtda:/sys/class/gpio/gpio203# cat direction 
out
root@mtda:/sys/class/gpio/gpio203# cat value 
1
```
and also checked that the `target off` command does reset the gpio back to zero:
```
root@mtda:/sys/class/gpio/gpio203# cat value 
0
```

chombourger, 2022-02-28:
looking at the logs, I wondered if @UKosale was using the CLI from the nanopi as the provided logs seem to suggest. I then tried on my end:
```
root@mtda:~# mtda-cli target on
# debug3: main.load_timeouts_config()
# debug3: main.load_timeouts_config: None
# debug3: main.start()
# debug3: main._target_off()
# debug3: main._target_off(): True
# debug3: main.storage_to_target()
# debug3: main._session_check(None)
# debug3: main._session_check: None
# debug3: main.storage_locked()
# debug3: main._session_check(None)
# debug3: main._session_check: None
# debug3: main._check_locked()
# debug3: main.target_owner()
# debug4: storage_locked(): no shared storage device
# debug3: main.storage_locked(): True
# debug1: storage_to_target(): shared storage is locked
# debug3: main.storage_to_target(): False
# debug3: main.target_on()
# debug3: main._session_check(Manchu)
# debug3: session Manchu: 300 seconds
# debug3: main._session_event(ACTIVE Manchu)
# debug3: main.notify(SESSION ACTIVE Manchu)
# debug3: main.notify: None
# debug3: main._session_event: None
# debug3: main._session_check: None
# debug3: main.target_status()
# debug3: main.target_status(): ???
# debug3: main.power_locked()
# debug3: main._session_check(Manchu)
# debug3: session Manchu: 300 seconds
# debug3: main._session_check: None
# debug3: main.power_locked(): True
# debug3: main.target_on(): False
```
and the gpio is indeed not toggled:
```
root@mtda:/sys/class/gpio/gpio203# cat value 
0
```

chombourger, 2022-02-28:
the work-around is to run the CLI as follows:
```
mtda@mtda:~$ mtda-cli -r localhost target on
# debug3: main.load_timeouts_config()
# debug3: main.load_timeouts_config: None
# debug3: main.start()
```
and voila, the gpio is properly toggled:
```
mtda@mtda:~$ cat /sys/class/gpio/gpio203/value 
1
```
It's a bit embarrassing... We do have #116 to separate the client and service code, this may help us with this issue as well.

@UKosale can you live with the work-around for now?

chombourger, 2022-02-28:
I should note that I am barely doing anything on the NanoPI / BeagleBone-Black nodes that are running the MTDA service, I am always interacting with the service remotely

chombourger, 2022-02-28:
Added the `bug` label. In the early days, we have had people use mtda without necessarily running a daemon/service. We have since added some features that are requiring a process to be sitting there and handling client requests. I am guessing that `mtda-cli` should assume it is running in client mode if the `-d` option isn't supplied

chombourger, 2022-02-28:
@UKosale feel free to look at the Pull Request and comment there.

UKosale, 2022-03-01:
I can live with workaround, thank you.

closed 2022-03-01 by UKosale

chombourger, 2022-03-01:
> I can live with workaround, thank you.

I have merged the fix proposed in #220, the work-around should no longer be necessary with HEAD or the upcoming 0.17 release

PR #220 (2022-03-01): mtda-cli: use 'localhost' as default remote
Clients should always connect to the daemon/service. Assume "localhost"
when no remote is specified on the command line instead of silently
failing.

Closes: #219
commit 7a8dc3aa (2022-03-01): mtda-cli: use 'localhost' as default remote

Clients should always connect to the daemon/service. Assume "localhost"
when no remote is specified on the command line instead of silently
failing.

Closes: #219