siemens/mtda #225: Multiple bugs in serial console driver

jan-kiszka, 2022-03-04:
I've fixed one,
```
diff --git a/mtda/console/serial.py b/mtda/console/serial.py
index f852965..f91e113 100644
--- a/mtda/console/serial.py
+++ b/mtda/console/serial.py
@@ -88,7 +88,7 @@ class SerialConsole(ConsoleInterface):
         self.mtda.debug(3, "console.serial.pending()")
 
         result = 0
-        if self.ser is not None:
+        if self.ser is not None and self.opened:
             result = self.ser.inWaiting()
 
         self.mtda.debug(3, "console.serial.pending(): %s" % str(result))
```
but there must be more:
```
resetting console to recover from read error (<class 'serial.serialutil.PortNotOpenError'>)...
Exception in thread console_rx:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/mtda/console/logger.py", line 314, in reader
    data = con.read(con.pending() or 1)
  File "/usr/lib/python3/dist-packages/mtda/console/serial.py", line 103, in read
    result = self.ser.read(n)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 557, in read
    buf = os.read(self.fd, size - len(read))
TypeError: an integer is required (got type NoneType)
```

Config is simply
```
[console]
variant=serial
port=/dev/ttyUSB0
```

jan-kiszka, 2022-03-06:
Turned out that the trigger for these problems was
```
diff --git a/mtda/console/logger.py b/mtda/console/logger.py
index 9a4b160..3fab37a 100644
--- a/mtda/console/logger.py
+++ b/mtda/console/logger.py
@@ -303,8 +303,9 @@ class ConsoleLogger:
 
         if self.power_controller is not None:
             self.power_controller.wait()
-            with self.rx_lock:
-                con.open()
+
+        with self.rx_lock:
+            con.open()
 
         while self.rx_alive is True:
             if self.power_controller is not None:
```
I'll send a PR. Later on also for adding a driver to my power control - the reason why there is no controller yet.

jan-kiszka, 2022-03-06:
BTW, those patterns
```
lock()
open()
unlock()
...
lock()
read()
unlock()
```
and also
```
lock()
close()
open()
unlock()
```
smell very funny. What is that rx_lock supposed to protect? Seems there are more sleeping issues.

jan-kiszka, 2022-03-06:
Seems pause/resume races against the reader thread. Those two are not properly synchronized against each other, nor is it clear what should happen to the read thread when pause is invoked. Can someone clarify this?

And:
```
diff --git a/mtda/console/logger.py b/mtda/console/logger.py
index 3fab37a..3457730 100644
--- a/mtda/console/logger.py
+++ b/mtda/console/logger.py
@@ -334,7 +334,7 @@ class ConsoleLogger:
                 retries = retries - 1
 
     def pause(self):
-        self.mtda.debug(3, "console.logger.resume()")
+        self.mtda.debug(3, "console.logger.pause()")
 
         with self.rx_lock:
             result = self.console.close()
```

jan-kiszka, 2022-03-06:
Confirmed: Commending out logger.pause/resume make things work, even with debugging turned off.

These methods look unfinished.

chombourger, 2022-03-06:
> Confirmed: Commenting out logger.pause/resume make things work, even with debugging turned off.
> 
> These methods look unfinished.

we should no longer have this race with #233 
would be great if you could give it a try

(tested here with and without `[power]` in my config)

commit 08154831 (2022-03-23): console/serial: check for port opened in pending/read/write

Closes: #225
Reported-by: Jan Kiszka <jan.kiszka@siemens.com>
closed 2022-03-23 by chombourger