siemens/mtda #232: Autostart stumbles over ttyUSB availability

jan-kiszka, 2022-03-06:
```
Mar 06 08:05:32 mtda systemd[1]: Started Multi-Tenant Device Access.
Mar 06 08:05:38 mtda mtda-cli[364]: Probe of the serial console failed!
Mar 06 08:05:38 mtda mtda-cli[364]: Failed to start the MTDA server!
Mar 06 08:05:38 mtda systemd[1]: mtda.service: Succeeded.
Mar 06 08:05:38 mtda systemd[1]: mtda.service: Consumed 3.853s CPU time.
```
Actually two issue: The serial console is not yet available - are we missing some dependency in the service? Starting manually later on works.

Second issue: The service reports success.

jan-kiszka, 2022-03-06:
At least the primary bug relates to the pause/resume race (see #225) and is resolved by not executing those functions. Autostart works now.

What remains is error-free termination under error conditions.

chombourger, 2022-03-06:
How about having a recipe shipping a systemd drop-in to make the systemd service wait for /dev/ttyUSB0 like done for /dev/sda with [meta-isar/recipes-conf/mass-storage/usb-ms-use-sda_0.6.bb](https://github.com/siemens/mtda/blob/master/meta-isar/recipes-conf/mass-storage/usb-ms-use-sda_0.6.bb)

jan-kiszka, 2022-03-06:
Well, it remains a local configuration decision. I'm using a USB-serial converter right now, but I could imagine connecting the RPi on-board UART directly with the target.

commit 5ccdbb71 (2022-03-23): console/serial: implement configure_systemd()

Generate a systemd dropin for mtda.service to start after we have
the device node specified in the "port" property available.

Closes: #232
closed 2022-03-23 by chombourger