siemens/mtda #237: Some web UI findings

jan-kiszka, 2022-03-06:
First of all, a nice bonbon on top of mtda-cli!

Some findings, still:
* high latency of outputs, interactive console usage is not really enjoyable - fixable?
* console size only 80x24, should scale with browser size
* clicking again on keyboard symbol should close the pop-up

chombourger, 2022-03-07:
thanks for reporting your findings. a lot remains to be done in web UI

chombourger, 2024-01-12:
upstream discussion around input/echo latency: https://github.com/xtermjs/xterm.js/issues/887

chombourger, 2024-01-12:
With https://github.com/siemens/mtda/pull/387 we are adding the xterm.js fit addon
seems to work for width, height seems to be stuck at 24, not sure why as the dimensions of the container (the `console` element) fit the Winbox window