siemens/mtda #250: Debugging start-up time of mtda-cli

jan-kiszka, 2022-10-13:
On my RPi4 - which is not really a slow device - I'm facing several seconds of delay between calling `mtda-cli` and getting the prompt. With debug=4, it takes almost 4 s to get the first output ("# debug3: main.load_remote_config()"). I also tried removing all drivers from the local config, but that had no impact.

Any suggestions how to debug this further?

chombourger, 2022-10-13:
is that with HEAD?

jan-kiszka, 2022-10-13:
Not yet, still on b92d55b7b2eb516633a6a3906440af8fd841611 (with Debian). I don't see "suspicious" commits since - do you expect differences?

chombourger, 2022-10-13:
me neither - was just curious to understand which "version" you were running. I will build and flash the Pi4 image

chombourger, 2022-12-01:
finally going to take a look

chombourger, 2022-12-09:
Reproduced on Pi4 using the following command:
```
root@mtda:/home/mtda# (echo start && mtda-cli -d -n 2>&1) | ts '[%Y-%m-%d %H:%M:%S]'
[2022-12-09 00:02:52] start
[2022-12-09 00:02:55] # debug3: main.load_timeouts_config()
[2022-12-09 00:02:55] # debug3: main.load_timeouts_config: None
[2022-12-09 00:02:55] # debug3: main._parse_script()
[2022-12-09 00:02:55] # debug3: main._parse_script(): scripts.power_on()
[2022-12-09 00:02:55] # debug3: main._parse_script()
[2022-12-09 00:02:55] # debug3: main._parse_script(): scripts.power_off()
[2022-12-09 00:02:55] # debug3: main.start()
```
and a minimalistic config:
```
root@mtda:/home/mtda# cat /etc/mtda/config
[main]
debug=3
```

chombourger, 2022-12-09:
reaching `main()` takes 3s alone as shown below:
```
root@mtda:/home/mtda# (echo start && mtda-cli -d -n 2>&1) | ts '[%Y-%m-%d %H:%M:%S]'
[2022-12-09 00:08:04] start
[2022-12-09 00:08:07] ### main
```

chombourger, 2022-12-09:
used `python3 -X importtime` to check import times and `self [us]` for mtda.main is 1476358 (~1.5s)

chombourger, 2022-12-09:
the overall import time is 2649476 (~2.5s)

fmoessbauer, 2023-10-03:
I debugged this as part of f61f2fabe7ecba521dfa5df2811778bc35f2c293, citing:

> The startup of the mtda service might take quite long on the nanopi-neo
board (~15sec). This is due to the loading of a ton of dependencies,
which cannot really be avoided.

The `mtda-cli` is affected as well, as it internally loads MTDA to get the (local) MTDA version. This can be speed up by generating the `.pyc` files (by executing it once as root), but it is still slow. IMHO, the whole import should be made more dynamic and ideally the dependency to MTDA itself could be avoided.