siemens/mtda #255: snapshot feature for the shared storage device

chombourger, 2022-12-05:
Today I was presented an issue that only occurs on first boot of the system. Needless to say that it would be a pain to have to rewrite your OS image each time you want to reboot into that initial state. I was consequently thinking that it could be useful to have our USB Mass Storage gadget expose a DM-managed block device instead of the actual block device initialized with `storage write`. Writes to the emulated USB device would go to a separate block device giving us the ability to revert to that initial state at any time.

jan-kiszka, 2022-12-05:
Sounds useful. Basically model that after qemu's capabilities: Boot up in snapshot mode, perform changes, powerdown, and then have the chance to commit them or to drop them again. I'm working like that with some of my qemu testing machines.

chombourger, 2022-12-05:
> Sounds useful. Basically model that after qemu's capabilities: Boot up in snapshot mode, perform changes, powerdown, and then have the chance to commit them or to drop them again. I'm working like that with some of my qemu testing machines.

good point about qemu. we could not add this capability to setups using USB Functions but also to our QEMU configuration so we end up with the same `storage` APIs (such as `storage commit` and `storage revert`)

closed 2025-07-12 by chombourger