siemens/mtda #267: : Storage write ends with exception

gylstorffq, 2023-01-11:
If i write a image to a usb drive with mtda-0.17-1 on a raspi4 with `mtda-cli storage write <image>` the process ends with the following exception:

```
  File "/usr/bin/mtda-cli", line 825, in <module>
    app.main()
  File "/usr/bin/mtda-cli", line 814, in main
    status = cmds[cmd](stuff)
  File "/usr/bin/mtda-cli", line 473, in storage_cmd
    return cmds[cmd](args)
  File "/usr/bin/mtda-cli", line 563, in storage_write
    status = self.agent.storage_write_image(
  File "/usr/lib/python3/dist-packages/mtda/client.py", line 269, in storage_write_image
    return self._storage_write(image, imgname, imgsize, callback)
  File "/usr/lib/python3/dist-packages/mtda/client.py", line 210, in _storage_write
    status = self.storage_close()
  File "/usr/lib/python3/dist-packages/mtda/client.py", line 148, in storage_close
    return self._impl.storage_close(self._session)
  File "/usr/lib/python3/dist-packages/zerorpc/core.py", line 278, in <lambda>
    return lambda *args, **kargs: self(method, *args, **kargs)
  File "/usr/lib/python3/dist-packages/zerorpc/core.py", line 270, in __call__
    return self._process_response(request_event, bufchan, timeout)
  File "/usr/lib/python3/dist-packages/zerorpc/core.py", line 227, in _process_response
    reply_event = bufchan.recv(timeout=timeout)
  File "/usr/lib/python3/dist-packages/zerorpc/channel.py", line 255, in recv
    event = self._input_queue.get(timeout=timeout)
  File "src/gevent/queue.py", line 334, in gevent._gevent_cqueue.Queue.get
  File "src/gevent/queue.py", line 349, in gevent._gevent_cqueue.Queue.get
  File "src/gevent/queue.py", line 326, in gevent._gevent_cqueue.Queue.__get_or_peek
    SimpleQueue = _PySimpleQueue
  File "src/gevent/_waiter.py", line 151, in gevent._gevent_c_waiter.Waiter.get
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
zerorpc.exceptions.LostRemote: Lost remote after 40s heartbeat

```

chombourger, 2023-01-11:
This suggests that the mtda daemon on the RPI4 did not respond to the ZeroRPC heartbeat.
Can we increase the debug level to 3 on the RPI side and check what the logs say (if anything)?

chombourger, 2023-01-11:
It may also be good to let us know:

- the compression format of your image
- the size of your image
- what would mtda be writing to on the RPI side? (did you configure the storage "driver" to write to a stick connected to the RPi which would then be exposed to the Device Under Test using the USB mass storage gadget?

it may also be interesting to check the kernel logs on the RPI

gylstorffq, 2023-01-12:
I tested it with no compression and xz in both cases the error occurs. The image is uncompressed ~8GB and compressed ~300MB. MTDA writes directly to a USB drive. Here is the config for storage:

```
[storage]
variant = usbf
file = /dev/sda

```

The image is written successfully to the disk.

gylstorffq, 2023-01-12:
[mtda_deamon_2023-01-09.log.gz](https://github.com/siemens/mtda/files/10401658/mtda_deamon_2023-01-09.log.gz)
Here is a log from the daemon.

chombourger, 2023-01-12:
Thank you so much for the additional details and for the logs.
I took only a quick glance and did not spot anything abnormal.
Will probably have to reproduce your setup (my primary device is the NanoPI for developing/testing MTDA, I however have a rpi4 that I use for some other projects but can use for spot tests)

chombourger, 2023-01-13:
For completeness, here's the config file used on my end:
```
[main]
debug=2

[power]
variant = shellcmd
on-cmd = /bin/true
off-cmd = /bin/true
check-on = /bin/false

[storage]
variant = usbf
file = /dev/sda

[console]
variant  = serial
port     = /dev/ttyUSB0
rate     = 115200
```
obviously the power commands are dummy ones (not really using this rpi4 with MTDA or with a DUT attached)

gylstorffq, 2023-02-13:
I tested it with `zst` compression - this works. So it leads to an error with gz,xz and raw images.

PR #371 (2023-11-22): fix some tricky bugs by setting flags at the proper timing and place
```sh

1. mtda-cli -r  $HOST_NAME storage write  1.0.1-installer.wic.xz
2. mtda-cli -r $HOST_NAME storage mount 3
3. mtda-cli -r $HOST_NAME storage update $image_path_in_installer.bmap $bmap_path
4. mtda-cli -r $HOST_NAME storage update $image_path_in_installer.xz $xz_path
5. mtda-cli -r  $HOST_NAME storage target
6. mtda-cli -r  $HOST_NAME target on
7. ...
8. mtda-cli -r  $HOST_NAME target off
9. mtda-cli -r  $HOST_NAME storage host
10. mtda-cli -r  $HOST_NAME storage write  1.0.2-installer.wic.xz
11. mtda-cli -r $HOST_NAME  [... cut]
closed 2023-11-22 by chombourger