siemens/mtda #269: issues with mtda-www debian package

bovi, 2023-01-16:
I notice that the mtda-www is not working out of the box with the debian package.

I notice that all mtda-www files are located under python3:

```sh
./lib/python3/dist-packages/mtda
./lib/python3/dist-packages/mtda/www.py
./lib/python3/dist-packages/mtda/templates
./lib/python3/dist-packages/mtda/templates/index.html
./lib/python3/dist-packages/mtda/assets
./lib/python3/dist-packages/mtda/assets/vnc.js
./lib/python3/dist-packages/mtda/assets/keyboard.css
./lib/python3/dist-packages/mtda/assets/keyboard.js
```

But all mtda files are located under python3.9:

```sh
./lib/python3.9/dist-packages/mtda
./lib/python3.9/dist-packages/mtda/keyboard
./lib/python3.9/dist-packages/mtda/keyboard/hid.py
./lib/python3.9/dist-packages/mtda/keyboard/__pycache__
./lib/python3.9/dist-packages/mtda/keyboard/__pycache__/qemu.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/keyboard/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/keyboard/__pycache__/hid.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/keyboard/__pycache__/controller.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/keyboard/qemu.py
./lib/python3.9/dist-packages/mtda/keyboard/controller.py
./lib/python3.9/dist-packages/mtda/keyboard/__init__.py
./lib/python3.9/dist-packages/mtda/scripts
./lib/python3.9/dist-packages/mtda/scripts/ipc227e.py
./lib/python3.9/dist-packages/mtda/scripts/__pycache__
./lib/python3.9/dist-packages/mtda/scripts/__pycache__/qemu.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/scripts/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/scripts/__pycache__/ipc227e.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/scripts/qemu.py
./lib/python3.9/dist-packages/mtda/scripts/__init__.py
./lib/python3.9/dist-packages/mtda/support
./lib/python3.9/dist-packages/mtda/support/__pycache__
./lib/python3.9/dist-packages/mtda/support/__pycache__/usb.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/support/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/support/usb.py
./lib/python3.9/dist-packages/mtda/support/__init__.py
./lib/python3.9/dist-packages/mtda/assistant
./lib/python3.9/dist-packages/mtda/assistant/homekit.py
./lib/python3.9/dist-packages/mtda/assistant/__pycache__
./lib/python3.9/dist-packages/mtda/assistant/__pycache__/assistant.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/assistant/__pycache__/homekit.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/assistant/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/assistant/assistant.py
./lib/python3.9/dist-packages/mtda/assistant/__init__.py
./lib/python3.9/dist-packages/mtda/__version__.py
./lib/python3.9/dist-packages/mtda/power
./lib/python3.9/dist-packages/mtda/power/usbrelay.py
./lib/python3.9/dist-packages/mtda/power/gpio.py
./lib/python3.9/dist-packages/mtda/power/__pycache__
./lib/python3.9/dist-packages/mtda/power/__pycache__/pduclient.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/docker.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/shellcmd.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/usbrelay.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/gpio.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/qemu.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/aviosys_8800.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/__pycache__/controller.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/power/aviosys_8800.py
./lib/python3.9/dist-packages/mtda/power/pduclient.py
./lib/python3.9/dist-packages/mtda/power/qemu.py
./lib/python3.9/dist-packages/mtda/power/shellcmd.py
./lib/python3.9/dist-packages/mtda/power/controller.py
[... 61 lines cut ...]
./lib/python3.9/dist-packages/mtda/discovery.py
./lib/python3.9/dist-packages/mtda/usb
./lib/python3.9/dist-packages/mtda/usb/switch.py
./lib/python3.9/dist-packages/mtda/usb/gpio.py
./lib/python3.9/dist-packages/mtda/usb/qemu_mass_storage.py
./lib/python3.9/dist-packages/mtda/usb/__pycache__
./lib/python3.9/dist-packages/mtda/usb/__pycache__/qemu_mass_storage.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/usb/__pycache__/gpio.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/usb/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/usb/__pycache__/switch.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/usb/__init__.py
./lib/python3.9/dist-packages/mtda/__init__.py
./lib/python3.9/dist-packages/mtda/video
./lib/python3.9/dist-packages/mtda/video/__pycache__
./lib/python3.9/dist-packages/mtda/video/__pycache__/qemu.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/video/__pycache__/__init__.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/video/__pycache__/controller.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/video/__pycache__/mjpg_streamer.cpython-39.pyc
./lib/python3.9/dist-packages/mtda/video/mjpg_streamer.py
./lib/python3.9/dist-packages/mtda/video/qemu.py
./lib/python3.9/dist-packages/mtda/video/controller.py
./lib/python3.9/dist-packages/mtda/video/__init__.py
```

Not sure if this is the cause as I don't understand the python / debian package management madness but to me it looks like the mtda-cli can't find the mtda.www for importing in the main.py. Btw. the ISAR has also issues with mtda-www as it doesn't package this part not at all.

chombourger, 2023-01-16:
Thank you for reporting. I am seeing the same issue. Looks like a regression

closed 2023-01-17 by chombourger

bovi, 2023-01-20:
@chombourger is there a release of this debian package planned?

chombourger, 2023-01-20:
> Reference

planning to push a test candidate tag today and ask some of the folks working with me to exercise that build so we get something out relatively soon (0.17 having several annoying issues)

PR #271 (2023-01-17): debian: install to /usr/lib/pythonX.Y/dist-packages/mtda
Closes: #269
commit 03c4629e (2023-01-17): debian: install to /usr/lib/pythonX.Y/dist-packages/mtda

Closes: #269