siemens/mtda #300: novnc bookworm package leads to the terminal of the mtda-www webview not working

kergos, 2023-06-27:
I built an image with the newly released debian bookworm image for the nanopi-neo.
After installing the mtda-www package, the terminal in the web-view is not loaded anymore.

This is because the current novnc bookworm package (https://packages.debian.org/bookworm/all/novnc/filelist) does not provide the required file `/usr/share/novnc/vendor/promise.js` (https://packages.debian.org/bullseye/all/novnc/filelist).

Apparently this was removed from novnc some time ago (https://github.com/novnc/noVNC/commit/890cff921d78835762fbdcfef862571c52cb3035)
This leads to the terminal in the web-view not properly loading. With the bullseye image it still works as intended.

chombourger, 2023-06-27:
Thanks for reporting @kergos

looks like `promise.js` gets included from `mtda/templates/index.html`
any idea how to fix this? should I look into this?

kergos, 2023-06-27:
If I understand it correctly  by looking at the basic example from novnc, (https://github.com/novnc/noVNC/blob/e8ad466e45a21598debe81824744d85ff2323c7b/vnc_lite.html) it seems like it could be changed. 

Basically `mtda/templates/index.html` would be needed to be rewritten somewhat to implement the new way to do it.
I currently lack the knowledge how to implement this properly and would need to read up on it, since it has been quite some time since I did anything web-related.

bovi, 2023-06-27:
@kergos I suggest we look at this and provide a merge request afterwards here. let's talk tomorrow.

kergos, 2023-08-02:
Fixed in #303

closed 2023-08-02 by kergos