siemens/mtda #313: update to latest isar

chombourger, 2023-08-04:
wanted to update our isar refspec to the latest in order to get the `jammy` distro (aka Ubuntu 22.04) supported
this caused our mutliconfig builds to fail:
```
$ ./kas-container build kas/debian/mtda-nanopi-all.yml
2023-08-04 07:35:52 - INFO     - kas 3.2.3 started
2023-08-04 07:35:52 - INFO     - /repo/kas/debian$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - /repo/kas/debian$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - /repo$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - Using /repo as root for repository mtda
2023-08-04 07:35:52 - INFO     - /work/isar$ git remote set-url origin https://github.com/ilbers/isar.git
2023-08-04 07:35:52 - INFO     - /work/isar$ git cat-file -t c63bb31b6f3adbdc12fedae38d58107982dced90
2023-08-04 07:35:52 - INFO     - Repository isar already contains c63bb31b6f3adbdc12fedae38d58107982dced90 as commit
2023-08-04 07:35:52 - INFO     - /repo$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - Using /repo as root for repository mtda
2023-08-04 07:35:52 - INFO     - /work/isar$ git status -s
2023-08-04 07:35:52 - INFO     - /work/isar$ git rev-parse --verify -q origin/c63bb31b6f3adbdc12fedae38d58107982dced90
2023-08-04 07:35:52 - INFO     - /work/isar$ git checkout -q c63bb31b6f3adbdc12fedae38d58107982dced90
2023-08-04 07:35:52 - INFO     - /repo$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - Using /repo as root for repository mtda
2023-08-04 07:35:52 - INFO     - /repo$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - Using /repo as root for repository mtda
2023-08-04 07:35:52 - INFO     - /work/isar$ /tmp/tmpog588a7q/get_bb_env /build
2023-08-04 07:35:52 - INFO     - /repo$ git rev-parse --show-toplevel
2023-08-04 07:35:52 - INFO     - Using /repo as root for repository mtda
2023-08-04 07:35:52 - INFO     - /build$ /work/isar/bitbake/bin/bitbake -c build mc:nanopi-neo:mtda-image mc:nanopi-r1:mtda-image
Loading cache: 100% |                                                                                                                                                                                                       | ETA:  --:--:--
Loaded 0 entries from dependency cache.
ERROR: /build/../work/isar/meta/recipes-core/isar-bootstrap/isar-bootstrap-target.bb: Error executing a python function in <code>:                                                                                          | ETA:  --:--:--

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 4, function: <module>
     0001:__anon_158__build____work_isar_meta_classes_base_bbclass(d)
     0002:__anon_162__build____work_isar_meta_classes_sstate_bbclass(d)
     0003:__anon_69__build____work_isar_meta_recipes_core_isar_bootstrap_isar_bootstrap_inc(d)
 *** 0004:__anon_30__build____work_isar_meta_classes_compat_bbclass(d)
File: '/build/../work/isar/meta/classes/compat.bbclass', lineno: 24, function: __anon_30__build____work_isar_meta_classes_compat_bbclass
     0020:    overrides = d.getVar('OVERRIDES').split(':')
     0021:
     0022:    localdata = bb.data.createCopy(d)
     0023:    new_overrides = [distro_arch] + [o for o in overrides if not o == package_arch]
 *** 0024:    localdata.setVar('OVERRIDES', ':'.join(new_overrides))
     0025:    isar_enable_compat_arch = localdata.getVar('ISAR_ENABLE_COMPAT_ARCH')
     0026:    compat_distro_arch = localdata.getVar('COMPAT_DISTRO_ARCH')
     0027:
     0028:    d.setVar('COMPAT_DISTRO_ARCH', compat_distro_arch)
Exception: TypeError: sequence item 0: expected str instance, NoneType found

ERROR: /build/../repo/meta-isar/recipes-multimedia/mjpg-streamer/mjpg-streamer_git.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 4, function: <module>
     0001:__anon_158__build____work_isar_meta_classes_base_bbclass(d)
     0002:__anon_162__build____work_isar_meta_classes_sstate_bbclass(d)
     0003:__anon_23__build____work_isar_meta_classes_sbuild_bbclass(d)
 *** 0004:__anon_30__build____work_isar_meta_classes_compat_bbclass(d)
     0005:__anon_24__build____work_isar_meta_classes_multiarch_bbclass(d)
     0006:__anon_30__build____work_isar_meta_classes_patch_bbclass(d)
     0007:__anon_105__build____work_isar_meta_classes_dpkg_base_bbclass(d)
File: '/build/../work/isar/meta/classes/compat.bbclass', lineno: 24, function: __anon_30__build____work_isar_meta_classes_compat_bbclass
     0020:    overrides = d.getVar('OVERRIDES').split(':')
     0021:
     0022:    localdata = bb.data.createCopy(d)
     0023:    new_overrides = [distro_arch] + [o for o in overrides if not o == package_arch]
 *** 0024:    localdata.setVar('OVERRIDES', ':'.join(new_overrides))
     0025:    isar_enable_compat_arch = localdata.getVar('ISAR_ENABLE_COMPAT_ARCH')
     0026:    compat_distro_arch = localdata.getVar('COMPAT_DISTRO_ARCH')
     0027:
     0028:    d.setVar('COMPAT_DISTRO_ARCH', compat_distro_arch)
Exception: TypeError: sequence item 0: expected str instance, NoneType found

ERROR: Parsing halted due to errors, see error messages above                                                                                                                                                                | ETA:  0:00:16
ERROR: /build/../work/isar/meta/recipes-devtools/sdk-files/sdk-files.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 4, function: <module>
     0001:__anon_158__build____work_isar_meta_classes_base_bbclass(d)
     0002:__anon_162__build____work_isar_meta_classes_sstate_bbclass(d)
     0003:__anon_23__build____work_isar_meta_classes_sbuild_bbclass(d)
 *** 0004:__anon_30__build____work_isar_meta_classes_compat_bbclass(d)
     0005:__anon_24__build____work_isar_meta_classes_multiarch_bbclass(d)
     0006:__anon_30__build____work_isar_meta_classes_patch_bbclass(d)
     0007:__anon_105__build____work_isar_meta_classes_dpkg_base_bbclass(d)
File: '/build/../work/isar/meta/classes/compat.bbclass', lineno: 24, function: __anon_30__build____work_isar_meta_classes_compat_bbclass
     0020:    overrides = d.getVar('OVERRIDES').split(':')
     0021:
     0022:    localdata = bb.data.createCopy(d)
     0023:    new_overrides = [distro_arch] + [o for o in overrides if not o == package_arch]
 *** 0024:    localdata.setVar('OVERRIDES', ':'.join(new_overrides))
     0025:    isar_enable_compat_arch = localdata.getVar('ISAR_ENABLE_COMPAT_ARCH')
     0026:    compat_distro_arch = localdata.getVar('COMPAT_DISTRO_ARCH')
     0027:
     0028:    d.setVar('COMPAT_DISTRO_ARCH', compat_distro_arch)
Exception: TypeError: sequence item 0: expected str instance, NoneType found

Summary: There were 4 ERROR messages, returning a non-zero exit code.
2023-08-04 07:35:53 - ERROR    - Command returned non-zero exit status 1
```

It appears that `DISTRO_ARCH` is `None` when parsing some of our recipes
I ran a `git bisect` on Isar and the offending commit is:
```
commit c63bb31b6f3adbdc12fedae38d58107982dced90 (HEAD)
Author: Adriaan Schmidt <adriaan.schmidt@siemens.com>
Date:   Fri Mar 3 11:20:43 2023 +0100

    add multiarch support

    This adds support for building packages for native and compat architectures
    to dpdk-base.bbclass.
    Thus, all package recipes automatically have a *-native and *-compat target,
    which can be used in DEPENDS/RDEPENDS definitions.
    Additionally those targets can be used in IMAGE_INSTALL, where they
    are automatically converted to install the correct debian package:
      foo-compat -> foo:${COMPAT_DISTRO_ARCH}
      foo-native -> foo:${HOST_ARCH}

    Note that the switch ISAR_ENABLE_COMPAT_ARCH still exist and controls
    addition of the compat architecture during bootstrapping.

    Signed-off-by: Adriaan Schmidt <adriaan.schmidt@siemens.com>
```

I compared our multiconfig files with those found in isar/meta-isar and was unable to find what we are possibly doing wrong

chombourger, 2023-08-04:
any idea @vj-kumar ?

vj-kumar, 2023-08-06:
> any idea @vj-kumar ?

Not at the moment. I will check this out.

vj-kumar, 2023-08-07:
> > > any idea @vj-kumar ?
> > 
> > 
> > Not at the moment. I will check this out.
> 
> Identified the scenario causing this. Debugging it.

The reason is DISTRO_ARCH is undefined, atleast for the base configuration in the multiconfig build. The reason is kas by default adds qemux86-64 machine as default. This is a non-existent machine in ISAR. It has only qemuamd64 and qemui386.
This will solve our issue. But, there should be a better way, kas based on build system could set the correct default.

```diff
diff --git a/kas/common/base.yml b/kas/common/base.yml
index bfd5c25..a155282 100644
--- a/kas/common/base.yml
+++ b/kas/common/base.yml
@@ -12,6 +12,7 @@
 header:
   version: 10
 
+machine: qemuamd64
 target: mtda-image
 
 repos:
@@ -20,7 +21,7 @@ repos:
       meta-isar:
   isar:
     url: https://github.com/ilbers/isar.git
-    refspec: 70c3dec0ea1d96dedc1b0aa96b5b7e64bec3939e
+    refspec: 1f500d59b02c34549f96b5ece93cd99884ab52bb
     layers:
       meta:
       meta-isar:
```

vj-kumar, 2023-08-07:
> > > > > any idea @vj-kumar ?
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > Not at the moment. I will check this out.
> > 
> > 
> > > 
> > 
> > 
> > > Identified the scenario causing this. Debugging it.
> > 
> > 
> > The reason is DISTRO_ARCH is undefined, atleast for the base configuration in the multiconfig build. The reason is kas by default adds qemux86-64 machine as default. This is a non-existent machine in ISAR. It has only qemuamd64 and qemui386.
> > This will solve our issue. But, there should be a better way, kas based on build system could set the correct default.
> 
> Let's proceed with your changes. I will raise a Pull Request for kas. Having your changes in would unblock our release activities
> 
> Thanks!
> 
> > ```diff
> > diff --git a/kas/common/base.yml b/kas/common/base.yml
> > 
> > index bfd5c25..a155282 100644
> > 
> > --- a/kas/common/base.yml
> > 
> > +++ b/kas/common/base.yml
> > 
> > @@ -12,6 +12,7 @@
> > 
> >  header:
> > 
> >    version: 10
> > 
> >  
> > 
> > +machine: qemuamd64
> > 
> >  target: mtda-image
> > 
> >  
> > 
> >  repos:
> > 
> > @@ -20,7 +21,7 @@ repos:
> > 
> >        meta-isar:
> > 
> >    isar:
> > 
> >      url: https://github.com/ilbers/isar.git
> > 
> > -    refspec: 70c3dec0ea1d96dedc1b0aa96b5b7e64bec3939e
> > 
> > +    refspec: 1f500d59b02c34549f96b5ece93cd99884ab52bb
> > 
> >      layers:
> > 
> >        meta:
> > 
> >        meta-isar:
> > ```

https://github.com/siemens/mtda/pull/317

commit 2de4d487 (2023-08-08): kas/common/base: set default machine

The default machine set by kas is qemux86-64. This machine is invalid in
isar. This results in DISTRO_ARCH being unset, which inturn causes build
issues in the latest ISAR, because it is used in a python anonymous
function in compat.bbclass.

Closes: #313
closed 2023-08-08 by chombourger