siemens/mtda #333: usbf storage configuration

erdemkahraman, 2023-09-11:
Hello developers, I'm playing with some features of MTDA. As a small feedback, I struggled a lot while configuring the HID keyboard.

To get back to the main topic;
I wanted to activate the storage feature and configured it as follows. 

```
[main]
debug=3
fuse=no

[console]
variant=serial
port=/dev/ttyUSB0
rate=115200

[monitor]
variant=serial
port=/dev/ttyUSB0
rate=115200

[keyboard]
variant=hid
device=/dev/hidg0

[storage]
variant=usbf
file=/dev/mmcblk1p3
```

And also I tried to connect this device to the target device, but got an error like that.
Then I changed file as device but the result was similar.

```
rock@rock-3a:~$ mtda-cli storage target
# debug3: main.load_timeouts_config()
# debug3: main.load_timeouts_config: None
# debug3: main.start()
failed to connect the shared storage device to the target!
```

Am I configuring It wrong, or missing sth (e.g. usb gadget kernel modules) or Is there any missing information about the documentation of configuring shared storage device.

Note: I am able to plug this mmcblk1p3 partition to the DUT via using g_mass_storage. But I wanted to do it via MTDA.

chombourger, 2023-09-12:
Hello,

Thanks for giving MTDA a try and providing feedback!

Is the mass storage gadget working if you remove the keyboard from your configuration ?

I am asking because USB controllers have different capabilities and may not support the same number of USB functions for composite 

Could you attach your full debug logs as well?

chombourger, 2023-09-12:
You don't seem to have a power controller. I will check how MTDA behaves. When we cannot power the target on/off from MTDA, we assume it is always ON but switching storage between the host and target may be done when we know the target is OFF

erdemkahraman, 2023-09-13:
Hello @chombourger 
First of all thanks for your fast reply 

> Hello,
>
> Thanks for giving MTDA a try and providing feedback!
>
> Is the mass storage gadget working if you remove the keyboard from your configuration ?
>
> I am asking because USB controllers have [...]

to the first question

Yes I tried to remove the keyboard configuration and also serial and monitor. Storage was standalone but It did not worked.
Debug level is at 3 and full debug logs that I have sent it already. I don't know if there is any more logs.

> You don't seem to have a power controller. I will check how MTDA behaves. When we cannot power the target on/off from MTDA, we assume it is always ON but switching storage between the host and target may be done when we know the target is OFF

And yes I have tried to add a power target configuration as dummy for MTDA but the result did not change

chombourger, 2023-09-16:
> Yes I tried to remove the keyboard configuration and also serial and monitor. Storage was standalone but It did not worked. Debug level is at 3 and full debug logs that I have sent it already. I don't know if there is any more logs.

well I am hoping that we would get more information from `journalctl -u mtda`. Logs you have shared are those from the client (`mtda-cli`), I'd like to see those from the server (`mtda-service`)

erdemkahraman, 2023-09-25:
Hi @chombourger 

I am attaching the logs.

[mtda-journal.log](https://github.com/siemens/mtda/files/12711664/mtda-journal.log)
[mtda-service.log](https://github.com/siemens/mtda/files/12711665/mtda-service.log)

chombourger, 2023-09-25:
Thank you. 

It appears that MTDA was told that the block device should appear as `/dev/sda` :
```
storage.usbf.probe(): /dev/sda not found!
```

Can you confirm you have a mass storage device attached to your system and that it shows up as /dev/sda?

chombourger, 2023-09-25:
Looking at your config file you had:
```
file=/dev/mmcblk1p3
```
I will check the code to understand why we are then looking for /dev/sda

chombourger, 2023-09-25:
What version (commit hash?) are you using by the way?

erdemkahraman, 2023-09-25:
> Thank you.
> 
> It appears that MTDA was told that the block device should appear as `/dev/sda` :
> 
> ```
> storage.usbf.probe(): /dev/sda not found!
> ```
> 
> Can you confirm you have a mass storage device attached to your system and that it shows up as /dev/sda?

There is no device called /dev/sda (Maybe I tried to run it with default conf with using device=/dev/sda or sth like that), I have a partition that is not mounted on my rootfs. I want to attach it to device under test and it is /dev/mmcblk1p3

And I am using the master branch the last commit is ```02d0e8c49547db1ec055903d8a8897386216ac65```

erdemkahraman, 2023-09-25:
Also my configuration is here 

```
rock@rock-3a:~$ cat /etc/mtda/config
[main]
debug=3

[console]
variant=serial
port=/dev/ttyUSB0
rate=115200

[storage]
variant=usbf
device=/dev/mmcblk1p3
```

```
rock@rock-3a:~$ mtda-cli storage target
# debug3: main.load_timeouts_config()
# debug3: main.load_timeouts_config: None
# debug3: main.start()
failed to connect the shared storage device to the target!
```

erdemkahraman, 2023-09-25:
> do you mind checking what happens if you replace device= with file= (in the [storage]) block?

Same behavior

erdemkahraman, 2023-09-25:
For example 

I am able to use mtda-cli storage with 

```mtda-cli storage write images/image.wic```

And I am attaching my device with ```g_mass_storage```

It is working fine with that way but only attaching with "mtda-cli storage target/host" is not working

erdemkahraman, 2023-09-26:
Hi @chombourger 

I guess, you have mentioned earlier, the target and storage features are binded why I said this?
Because I used the power feature via using shellcmd variant and then I reconfigured the storage feature,

```
[main]
debug=3

[storage]
variant=usbf
file=/dev/mmcblk1p3

[power]
variant = shellcmd
on-cmd = /bin/true
off-cmd = /bin/true
check-on = /bin/false
```

And then I manually added the ```usb_f_mass_storage``` kernel module and then ```mtda-cli storage target/host``` worked.

But. The storage feature is not working ""standalone"", but dummy power settings fits me no problem.

Thanks a lot for your effort. I am closing the issue.

closed 2023-09-26 by erdemkahraman

chombourger, 2023-09-26:
thank you for the update @erdemkahraman 
feel free to provide feedback or recommend changes as you use mtda