siemens/mtda #354: Error when I am using different features at the same time.

erdemkahraman, 2023-10-17:
Hi, I was trying to use the mtda with these kind of features

- power (shellcmd)
- storage (usbf)
- console (serial)
- monitor (serial)

I am controlling the pdu over a script this is mocked as like that

```
#!/bin/bash

power_on(){
    sudo echo "power_on" > /home/q/power
}

power_off(){
    sudo echo "power_off" > /home/q/power
}

power_check(){
    status=$(cat /home/q/power)

    if [ $status == "power_on" ]; then
        echo "$status return 0"
	    return 0
    else
        echo "$status return 1"
	    return 1
    fi
}

if [ $# -ne 1 ]; then
    echo "Usage: $0 [on|off|check]"
    exit 1
fi

case "$1" in
    on)
        power_on
        ;;
    off)
        power_off
        ;;
    check)
        power_check
        ;;
    *)
        echo "Invalid argument: $1"
        echo "Usage: $0 [on|off|check]"
        exit 1
        ;;
esac
```

And my configuration is here;

```
[main]
debug=3

[console]
variant=serial
port=/dev/ttyUSB0
rate=115200

[monitor]
variant=serial
port=/dev/ttyUSB0
rate=115200

[storage]
variant=usbf
file=/dev/mmcblk1p3

[power]
variant=shellcmd
on-cmd=/home/q/power.sh on
off-cmd=/home/q/power.sh off
check-on=/home/q/power.sh check

```

I am trying to use the console with ```mtda-cli console send "dummy\n"```
And the isssue is; console feature is not working. There is not any bytes on the serial port.

When I remove the power setting on the configuration, console is working well.

Also I was had an issue [!33](https://github.com/siemens/mtda/issues/333) and storage feature was not feature. After this configuration (that I mentioned earlier on this issue page), The storage and power setting is working. But only console feature is not working. Is there any tricky way to solve it and, Why these features are dependent each other?

erdemkahraman, 2023-10-17:
For example in the logs

Here;

```
Oct 18 02:10:07 raspberrypi mtda-service[5506]: # debug3: main.console_locked(): False
Oct 18 02:10:07 raspberrypi mtda-service[5506]: # debug3: console.serial.write(data=b'erdem\n')
Oct 18 02:10:07 raspberrypi mtda-service[5506]: # debug3: console.serial.write(): None
Oct 18 02:10:07 raspberrypi mtda-service[5506]: # debug3: main.console_send(): None
Oct 18 02:10:13 raspberrypi mtda-service[5506]: # debug3: main.console_send()
```
It says console.serial.write(): None (with setted all configurations "power, storage, console, monitor")

And also here (power option disabled and "worked" case);

```
Oct 18 02:14:12 raspberrypi mtda-service[6087]: # debug3: main.console_locked(): False
Oct 18 02:14:12 raspberrypi mtda-service[6087]: # debug3: console.serial.write(data=b'last-guy\n')
Oct 18 02:14:12 raspberrypi mtda-service[6087]: # debug3: console.serial.write(): 9
Oct 18 02:14:12 raspberrypi mtda-service[6087]: # debug3: main.console_send(): None
Oct 18 02:14:14 raspberrypi mtda-service[6087]: # debug3: main._session_check(None)

```

chombourger, 2023-10-18:
Well serial and monitor cannot use the same UART. This monitor feature is meant to be used on systems providing some sort of target control interface. QEMU is such an example with this interface allowing the entry of arbitrary QEMU control commands

erdemkahraman, 2023-10-18:
Allright, I have removed the monitor feature on the config. Console is still not working with power(shellcmd) configuration.

chombourger, 2023-10-18:
writes only happen if the target is on: is this the case?

`_target_on()` calls `on()` from the selected power controller and only if that method returned True then we call `console_logger.resume()` which will get the specified serial port opened

`SerialConsole.write()` does nothing and returns `None` if the serial port was not opened

chombourger, 2023-10-18:
from your logs, write() returns None when target is OFF (as expected) and returns > 0 when it is ON
things look ok to me

erdemkahraman, 2023-10-24:
I guess there is a misunderstanding on the shellcmd's README 

I fixed this issue by using this script with these return codes.

```
power_on(){
    sudo pypwrctrl on -d 1
    local power_status=$(sudo pypwrctrl -d show | grep "Nr. 1" | awk '{print $NF}' | tr -d '()')
    if [ "$power_status" = "on" ]; then
        return 0
    else
        return 1
    fi
}

power_off(){
    sudo pypwrctrl off -d 1
    local power_status=$(sudo pypwrctrl -d show | grep "Nr. 1" | awk '{print $NF}' | tr -d '()')
    if [ "$power_status" = "off" ]; then
        return 0
    else
        return 1
    fi
}

power_check(){
    local power_status=$(sudo pypwrctrl -d show | grep "Nr. 1" | awk '{print $NF}' | tr -d '()')
    if [ "$power_status" = "off" ]; then
        return 1
    else
        return 0
    fi
}
```

And the README is here;

```
shellcmd driver settings
The shellcmd driver may be used to control power switch with custom shell commands, e.g. curl requests:

on-cmd: string [required]
Power-on shell command. The return code should be 1 on success.

off-cmd: string [required]
Power-off shell command. The return code should be 1 on success.

check-on: string [required]
Shell command to check the power state. Should return 0 if power is on, 1 if it is off. Any other return code is interpreted as error.
```

It says at on-cmd command, return code 1 on success; but AFAIK return 1 code is an error code, but I neverminded it and configured success case as return 1 (and also the other configurations set as similar).

After checking the logs you said that if the target is off mtda will not try to send sth to the console. Yes you are true, Check-on command's return code was wrong. After changing this return codes, earlier I've shared. The issue is fixed.

chombourger, 2023-10-24:
thanks for the update. would you raise a PR to clarify the documentation or is that not needed?

erdemkahraman, 2023-10-24:
Sure I'm ready for contribute😊

chombourger, 2024-01-22:
can we close this?

closed 2024-02-13 by erdemkahraman