siemens/mtda #366: storage_update doesn't work

DingMeng-Siemens, 2023-11-14:
_storage_update_ doesn't work  because:
1. the storage is not open before call __storage_write_ in _storage_update_
2. the __storage_write_ is removed from the latest commitment, but _storage_update_ still use this function

By the way, we have a use case: user just wants to flash the installer to the shared storage once, and copy some license files or different target images to installer frequently (there is a default target image in the /media partition which is used for user to copy their own images  in installer). But I don't see any solution for that situation.

bovi, 2023-11-15:
@chombourger it seems here the change was made: https://github.com/siemens/mtda/commit/0ccb5c6ba146d317940cdee1d5fcd48f58b9d517#diff-6a8cf01935ea3624a4ca3c07b29a8700aac7da76a1ce4d2932e6df875087eddaL179 and currently it is still used here: https://github.com/siemens/mtda/blob/master/mtda/client.py#L199

DingMeng-Siemens, 2023-11-17:
Already fix this,  I will make pull request in the next few days. cc @bovi

closed 2023-11-20 by DingMeng-Siemens