siemens/mtda #498: On www physical keyboard special keys support is missing

shikl3x7, 2025-07-16:
on v0.39 when Enter or any Special Keys are pressed over the Video Stream Window, it's not recognized.
Logs of MTDA when Enter Key is pressed.
```console
mtda@mtda-003:~$ sudo journalctl -u mtda -f
[sudo] password for mtda:
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: storage.helpers.image.tell()
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: storage.helpers.image.tell(): None
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: main.storage_status(): ('TARGET', False, 0)
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: main.video_format()
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: main.video_format(): MJPG
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: main.video_url(host=134.86.254.110, opts=None)
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: video.mjpg_streamer.url(host='134.86.254.110')
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: video.mjpg_streamer.url(): http://134.86.254.110:8080/?action=stream
Jul 16 06:08:49 mtda-003 mtda-service[434]: # debug3: main.video_url(): http://134.86.254.110:8080/?action=stream
Jul 16 06:13:52 mtda-003 mtda-service[434]: # debug2: device will be powered down in 86400 seconds
Jul 16 07:27:22 mtda-003 mtda-service[434]: # debug3: main.mouse_move()
Jul 16 07:27:22 mtda-003 mtda-service[434]: # debug3: mouse.hid.move(0.3897058823529412, 0.4830729166666667, 1)
Jul 16 07:27:22 mtda-003 mtda-service[434]: # debug3: mouse.hid._write()
Jul 16 07:27:23 mtda-003 mtda-service[434]: # debug3: mouse.hid._write(): 0
Jul 16 07:27:23 mtda-003 mtda-service[434]: # debug3: mouse.hid.move(): 0
Jul 16 07:27:23 mtda-003 mtda-service[434]: # debug3: main.mouse_move(): None
Jul 16 07:27:23 mtda-003 mtda-service[434]: # debug3: main.mouse_move()
Jul 16 07:27:23 mtda-003 mtda-service[434]: # debug3: mouse.hid.move(0.3897058823529412, 0.4830729166666667, 0)
Jul 16 07:27:23 mtda-003 mtda-service[434]: # debug3: mouse.hid._write()
Jul 16 07:27:24 mtda-003 mtda-service[434]: # debug3: mouse.hid._write(): 0
Jul 16 07:27:24 mtda-003 mtda-service[434]: # debug3: mouse.hid.move(): 0
Jul 16 07:27:24 mtda-003 mtda-service[434]: # debug3: main.mouse_move(): None
Jul 16 07:27:24 mtda-003 mtda-service[434]: # debug3: main.keyboard_press()
Jul 16 07:27:24 mtda-003 mtda-service[434]: # debug3: keyboard.hid.press()
Jul 16 07:27:24 mtda-003 mtda-service[434]: # debug3: main.keyboard_press(): None
Jul 16 07:27:36 mtda-003 mtda-service[434]: # debug3: main.keyboard_press()
Jul 16 07:27:36 mtda-003 mtda-service[434]: # debug3: keyboard.hid.press()
Jul 16 07:27:36 mtda-003 mtda-service[434]: # debug3: main.keyboard_press(): None

```

chombourger, 2025-07-16:
Noticed that the following did not work in FF (at least for me):

- Enter
- Backspace
- /

chombourger, 2025-07-16:
You may want to check the Browser console (using the Web developer tools). We have several `console.log` prints in `index.html`

shikl3x7, 2025-07-16:
Looks like that would not resolve, checking in main now